blueman-project/blueman #1071: blueman-manager display dpi issues

532910, 2019-07-13:
blueman: 2.1.1-1
BlueZ: 5.50-1
Distribution: debian/sid
Desktop environment: enlightenment 0.22.4-2

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Sometimes it looks oversized and I'm unable to press anywhere.
[image]
(Changeing the window size makes it look right)

Help->Help always looks undersized:
[image]

All other apps (blueman-adapters, blueman-assistant, blueman-services) looks fine.

cschramm, 2019-07-23:
Hey, I just tried enlightenment 0.22.4-2 on my system and could not replicate that issue with the device window. Does it happen at any specific window size or anything?

532910, 2019-07-23:
Hmm. I've found the root of the problem. It happens when E tries to set window size (via Window Remembers).
When size is not remembered all windows look fine.

infirit, 2019-07-23:
@532910 We use Gtk as gui toolkit to deal with this for us and it appears the interaction with E and position/size causes problems. Not sure how much we can do about this :thinking:

cschramm, 2019-07-23:
If we're the only (GTK+ 3) app, we're probably doing something fancy that triggers it. I'll play some more and try to replicate it.

cschramm, 2019-07-27:
> It happens when E tries to set window size (via Window Remembers).

I'm still not able to replicate that. Can you guide me how to do that? Do I need any specific settings?

532910, 2019-08-06:
1. launch blueman-manager
2. Window -> Maximize -> Maximize
3. Window -> Remember
Identifiers -> set `Window name` chekbox
Properties -> set `Size` chckbox

cschramm, 2019-09-04:
I followed your steps but unfortunately still could not reproduce the issue. Please provide the output of `dconf read /org/blueman/general/window-properties` (you need `dconf-cli`), maybe the values are relevant somehow.

532910, 2019-09-05:
```
% dconf read /org/blueman/general/window-properties
[1200, 1600, 0, 0]
```
And monitor is 1200x1600 and the window that I view is 1200x1600. But looks like there is another size that is not 1200x1600 as I can move (with Alt-Leftmouse) this 1200x1600 window only by some area of the window.

When I change the window size (with Alt-Middlemouse) all sizes become equal and picture becomes look as it should.

532910, 2019-09-05:
Try to check if `Remember` settings are saved.

Try to change size and `Maximize` again after it.

The issue occurs only on blueman-manager launch.

532910, 2019-09-11:
I've just seen the same problem with KiCad, so it's not blueman-manager issue.

closed 2019-09-11 by 532910