blueman-project/blueman #1072: Device not pairing. Error: org.freedesktop.DBus.Error.NoReply

P0linom, 2019-07-16:
blueman: 2.0.5
BlueZ: 5.50
Distribution:  Ubuntu 18.04
Desktop environment: Gnome

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

When using **blueman-applet** ->

`UpdatePowerState (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:159)
off False 
foff False 
on True 
current state True 
new state True 
opacity 174
opacity 174
org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.`

When using **bluetoothctl** ->

```
- [bluetooth]# agent on
- Agent is already registered
- [bluetooth]# devices
- Device 23:B3:20:BE:E2:F7 NF-01
- [bluetooth]# pair 23:B3:20:BE:E2:F7
- Attempting to pair with 23:B3:20:BE:E2:F7
- Failed to pair: org.bluez.Error.AuthenticationCanceled
```

cschramm, 2019-07-22:
Unfortunately there's no traceback in your output. What action triggers the error? Pairing? Do you run `blueman-applet` and `bluetoothctl` for the same user?

infirit, 2019-07-23:
Follow up question, you should see a notification or dialog asking to confirm the pin. Do you see that?

gothaf, 2019-09-02:
Logitech K380 keyboard pairing fails through blueman-manager. No pin is shown. Keyboard pairs succesfully using bluetoothctl pair command.

closed 2019-11-08 by stale[bot]