blueman-project/blueman #1073: Remote control via bluetooth headset

nmschulte, 2019-07-16:
```
blueman: 2.1.1
BlueZ: 5.50
Distribution: Debian Bullseye/Sid
Desktop environment: X.Org, SDDM, i3
```

I use Blueman with i3 window manager on a Debian system.  I'm wondering how I can leverage the controls (playback, volume, etc.) on my bluetooth headset from my host.

I believe this is possible, given this document, but I must have something misconfigured or missing.  https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/media-api.txt
Monitoring d-bus doesn't give any clues at the moment.

I use A2DP sound profile; the new "Auto connect profiles" feature (since 2.0.8) are very nice, thank you!

cschramm, 2019-07-22:
Here is what I see when I connect such a headset:

The  bluetooth device provides the services `0000110c-0000-1000-8000-00805f9b34fb` and `0000110e-0000-1000-8000-00805f9b34fb` for remote control.

When I connect `uinput` picks it up as a new device: `input: A0:E9:DB:00:32:86 as /devices/virtual/input/input31`.

`xinput` shows a device for that:
`A0:E9:DB:00:32:86                       	id=21	[slave  keyboard (3)]`

`xinput test-xi2 21` shows key presses:
```
A0:E9:DB:00:32:86                       	id=21	[slave  keyboard (3)]
	Reporting 1 classes:
		Class originated from: 21. Type: XIKeyClass
		Keycodes supported: 248

EVENT type 13 (RawKeyPress)
    device: 21 (21)
    detail: 209
    valuators:

EVENT type 14 (RawKeyRelease)
    device: 21 (21)
    detail: 209
    valuators:
```

I'm not sure who's supposed to pick them up and how they get delivered to media applications. It just works on my system (MATE desktop). You probably need to bind the media keys manually, see e.g. https://wiki.debian.org/Keyboard/MultimediaKeys, https://www.reddit.com/r/i3wm/comments/2bhhog/mediakeys_spotify_for_noobs/, ... Note that this is independent of the fact that it's a bluetooth headset and would be the same for e.g. a USB keyboard with multimedia keys.

nmschulte, 2019-07-22:
@cschramm thank you!  I was able to make this work via `xmodmap`.  As well `playerctl` from the reddit discussion is very handy.

```
$ bluetoothctl info 'CC:98:8B:94:8F:EA'`
Device CC:98:8B:94:8F:EA (public)
	Name: WH-1000XM3
	Alias: WH-1000XM3
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (0000fe03-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (5b833e05-6bc7-4802-8e9a-723ceca4bd8f)
	UUID: Vendor specific           (5b833e06-6bc7-4802-8e9a-723ceca4bd8f)
	UUID: Vendor specific           (69a7f243-e52f-4443-a7f9-cb4d053c74d6)
	UUID: Vendor specific           (7b265b0e-2232-4d45-bef4-bb8ae62f813d)
	UUID: Vendor specific           (81c2e72a-0591-443e-a1ff-05f988593351)
	UUID: Vendor specific           (91c10d9c-aaef-42bd-b6d6-8a648c19213d)
	UUID: Vendor specific           (931c7e8a-540f-4686-b798-e8df0a2ad9f7)
	UUID: Vendor specific           (96cc203e-5068-46ad-b32d-e316f5e069ba)
	UUID: Vendor specific           (b9b213ce-eeab-49e4-8fd9-aa478ed1b26b)
	UUID: Vendor specific           (f8d1fbe4-7966-4334-8024-ff96c9330e15)
	UUID: Vendor specific           (fe59bfa8-7fe3-4a05-9d94-99fadc69faff)
	Modalias: usb:v054Cp0CD3d0411
```
----
```
# dmesg | grep input
input: CC:98:8B:94:8F:EA as /devices/virtual/input/input33
```
```
$ xinput list 'CC:98:8B:94:8F:EA'`
CC:98:8B:94:8F:EA                       	id=21	[slave  keyboard (3)]
	Reporting 1 classes:
		Class originated from: 21. Type: XIKeyClass
		Keycodes supported: 248

```
```
$ cat ~/.xmodmaprc
keycode 171 = XF86AudioNext
keycode 173 = XF86AudioPrev
keycode 208 = XF86AudioPlay
keycode 209 = XF86AudioPause
```

closed 2019-07-22 by nmschulte

infirit, 2019-07-23:
Just to add a little for future readers. The media dbus api you linked to is to control a remote playback device. For example a phone playing through your computers speakers.