blueman-project/blueman #1075: Add ability to trust devices by name

RodrigoLeiteF, 2019-07-17:
blueman: 2.1
BlueZ: 5.50-6
Distribution: Antergos
Desktop environment: i3wm

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Currently Blueman-Manager will only attempt to automatically pair with devices that it can recognize the Mac Address as far as I can tell. This is fine with the exception being the Steam Controller, who will always report the name "SteamController" but different Mac Addresses (with the first pair being CE every time, apparently), meaning the user has to manually pair the device every single time.

The only solution I can think of would be to always try to pair with devices named "SteamController". Is there a known workaround for this?

cschramm, 2019-07-21:
Hi Rodrigo, I cannot find any reference for that "feature" of the steam controller. Can you provide one?

RodrigoLeiteF, 2019-09-19:
Hey. Sorry, I completely forgot about this issue.

Turns out the Steam Controller only reports a different address if you turn it on by holding X+Steam. If you simply pair it using that method and subsequently only press the Steam button, it will report as the same device as before.

closed 2019-09-19 by RodrigoLeiteF