blueman-project/blueman #1076: Device not pairing. DBusAuthenticationFailedError

nolyn, 2019-07-19:
blueman: 2.0.5-1ubuntu1
BlueZ: 5.48-0ubuntu3.1
Distribution: Ubuntu 18.04
Desktop environment: Lightdm

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

We are using a Ubuntu 18.04 live system, which is slightly modified. Bluetooth works on my laptop, but we are having trouble when using a USB bluetooth dongle (AVM Bluefritz USB) on a desktop computer. 
Devices can be searched with blueman, but only mac addresses are shown, the names do not show up. When attempting to pair to a device we get a DBusAuthenticationFailedError. There are almost no messages from blueman-applet:
```
on_adapter_property_changed (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 1 
_________
on_adapter_property_changed (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 0 
_________
RequestPinCode (/usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py:162)
Agent.RequestPinCode 
```
No popup appears, to enter the pin. 

Strangely the following works: 
```
bluetoothctl
scan on
```
After a while the names of the bluetooth devices appear and now also show up in the blueman-applet. Now pairing works, we get a pop and can enter the pin. I can then connect a serial port to my device and it works as it should. 

Also something that doesn't work is disconnecting the serial port again, though I don't strictly need this. 
Disconnecting the serial port:
```
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_00_16_53_52_E0_AD 
_________
Destroy (/usr/lib/python3/dist-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_00_16_53_52_E0_AD 
_________
```
It is still connected however. Disconnecting the whole device:
```
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
on_device_disconnect (/usr/lib/python3/dist-packages/blueman/plugins/applet/SerialManager.py:135)
Disconnecting /dev/rfcomm0 
_________
on_device_disconnect (/usr/lib/python3/dist-packages/blueman/plugins/applet/SerialManager.py:135)
Failed to disconnect /dev/rfcomm0 
_________
on_device_disconnect (/usr/lib/python3/dist-packages/blueman/plugins/applet/SerialManager.py:135)
org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 36, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python3/dist-packages/blueman/plugins/mechanism/Rfcomm.py", line 26, in close_rfcomm
    if command in line:
TypeError: a bytes-like object is required, not 'str'
 
_________
__del__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_00_16_53_52_E0_AD 
_________
Destroy (/usr/lib/python3/dist-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_00_16_53_52_E0_AD 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_00_16_53_52_E0_AD 
_________
Destroy (/usr/lib/python3/dist-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_00_16_53_52_E0_AD 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_00_16_53_52_E0_AD 
_________
Destroy (/usr/lib/python3/dist-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_00_16_53_52_E0_AD 
```

cschramm, 2019-07-22:
If the device was discovered with blueman, you do not see its name and pairing fails with `DBusAuthenticationFailedError`.
If it got discovered with bluetoothctl, you do see its name and pairing works.
Correct?

That's weird as both discovery methods should totally have the same effect. Seeing only addresses and no names is what I expect for BLE. I'll see what I can find out about bluetoothctl's magic.

Regarding the `DBusAuthenticationFailedError`, bluethoothd's log output (probably in `daemon.log` on your system) might give you more information on what actually goes wrong. Otherwise, you can start `bluetoothd` manually with the `-d` and `-n` flags to see debug output.

nolyn, 2019-07-22:
Here is the output of bluetoothd -d -n:
https://pastebin.com/fjQtykeq

This time I was not able at all to connect. Also not after using bluetoothctl. Could it remember the failed connection attempt with blueman? I'll test some more ...

cschramm, 2019-07-22:
```
bluetoothd[10854]: src/device.c:new_auth() Requesting agent authentication for 00:16:53:52:E0:AD
bluetoothd[10854]: No agent available for request type 0
```

No agent was registered, i.e. probably neither blueman-applet was running nor did you register an agent with the `agent` command in `bluetoothctl`.

Regarding pre-pairing name resolution, I just played with `bluetoothctl` and always only got MAC-address like placeholder names (BlueZ 5.50, Linux 4.19.37).

nolyn, 2019-07-22:
I tried starting bluetoothd -n -d with sudo instead of root, but got the same result. One more thing I noticed: I accidentally tried to pair another device then the one I'm testing using blueman and I immediately got the popup, also afterwards it knew the name. While my other device was still shown with the mac address and I couldn't pair it until doing the bluetoothctl trick

nolyn, 2019-07-22:
I noticed something that might be helpful in the journal. This is what I get when I try to connect with blueman.
```
Jul 22 15:31:50 ubuntu bluetoothd[1240]: Agent /org/blueman/agent/global replied with an error: org.freedesktop.DBus.Python.KeyError, Traceback (most recent call last):
                                           File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
                                             retval = candidate_method(self, *args, **keywords)
                                           File "/usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py", line 167, in RequestPinCode
                                             self.ask_passkey(device, dialog_msg, notify_msg, False, self.notifications, ok, err)
                                           File "/usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py", line 122, in ask_passkey
                                             alias = self.get_device_alias(device_path)
                                           File "/usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py", line 95, in get_device_alias
                                             name = props["Name"]
                                         KeyError: 'Name'

```

cschramm, 2019-07-22:
That issue is fixed in 2.1 and related to some rare legacy devices that do not provide a name (and probably a change in how BlueZ handles that).

closed 2019-07-22 by cschramm

infirit, 2019-07-23:
The fix for that should be simple by adding a default name property to main.Device. The question is do we want to keep maintaining fixes for 2.0, let alone do a release. Maybe we can do a commit in 2.0 for distros to pick from git?

@nolyn Ill post a patch here for you to try and at least get it working.

nolyn, 2019-07-23:
Thanks! After @cschramm mentioned it is fixed in 2.1 I tried to find the corresponding bug and found this one https://github.com/blueman-project/blueman/issues/366. I tried to patch my system with #368, but there was some other error. Can check again what it was if it will help.
And I'm happy to try your patch :)

cschramm, 2019-07-23:
> The question is do we want to keep maintaining fixes for 2.0, let alone do a release. Maybe we can do a commit in 2.0 for distros to pick from git?

Debian / Ubuntu unfortunately would not pick them up unless it's security related, so I don't see a reason to do that...

infirit, 2019-07-23:
@nolyn two patches, the first possibly breaks compatibility with bluez 4 and the second just adds a default value for the Name property. The first is the correct way to get a name in bluez5 so start with that one.

```diff
diff --git a/blueman/main/applet/BluezAgent.py b/blueman/main/applet/BluezAgent.py
index 15c5b544..12b5759e 100644
--- a/blueman/main/applet/BluezAgent.py
+++ b/blueman/main/applet/BluezAgent.py
@@ -92,7 +92,7 @@ class CommonAgent(_GObjectAgent, Agent, GObject.GObject):
         device = Bluez.Device(device_path)
         props = device.get_properties()
         address = props["Address"]
-        name = props["Name"]
+        name = props["Alias"]
         alias = address
         if name:
             alias = "<b>%s</b> (%s)" % (cgi.escape(name), address)
```

```diff
diff --git a/blueman/main/Device.py b/blueman/main/Device.py
index 5b70fb70..138a1deb 100644
--- a/blueman/main/Device.py
+++ b/blueman/main/Device.py
@@ -38,6 +38,7 @@ class Device(GObject.GObject):
         #set fallback icon, fixes lp:#327718
         self.Device.Icon = "blueman"
         self.Device.Class = "unknown"
+        self.Device.Name = "No name"
         self.Device.Appearance = 0
 
         self.Valid = True
```

nolyn, 2019-07-24:
@infirit the first patch did the trick. Didn't try the second one. Once the device got connected, the name showed up as well. Thank so much!
I could not figure out how to the rfcomm device again once it was set up. I tried in the blueman GUI and also on cmdline with rfcomm release and bluetoothctl remove. However the device only disappears once I kill blueman-rfcomm-watcher. Did I miss something there? A matter of curiosity, I don't expect to have to disconnect the serial port once it was set up for my use cases.

cschramm, 2019-07-24:
That's probably due to a bug in the code that should kill the watcher fixed in https://github.com/blueman-project/blueman/pull/939/commits/fb317fc16ac3f7a5958092109b8c6d3252605fcd (2.0.7).

infirit, 2019-07-24:
@nolyn unfortunately serial/rfcomm was pretty broken. As @cschramm said lots of issues got fixed in 2.0.7 but I am not sure where you can pickup 2.0.8/7 for 18.04. You could try blueman from eoan.

cschramm, 2019-07-24:
All available 2.0.8 deb packages depend on Python 3.7 so they are not installable on bionic as it has 3.6 but you could download the source package and build with dpkg-buildpackage if you're adventurous.