blueman-project/blueman #1077: issue: Shall bluetooth get enabled automatically

Defossez, 2019-07-22:
Hi,

Brand new Linux Mint 19.1 install on Dell Precision M4800 laptop.
Internal Bluetooth mini-card disabled. External USB Bluetooth dongle used.

Default Blueberry is installed and that works perfectly. Connection to Bluetooth devices, file transfer, ... all works OK.
Needed for proprietary reasons more control than the simple Blueberry allows.
Removed Blueberry (sudo apt remove --purgeblueberry) and after reboot installed Blueman (sudo apt install blueman). Both operations happen without errors or warnings.
After reboot tried Blueman by clicking icon in taskbar. This brings a lot more options than available in Blueberry.
Top of pop-up menu says: Turn Bluetooth off --> Thus I expect it's running fine.
When clicking any other menu selection a small screen pop-up appears with the question:
 "Shall Bluetooth get enabled automatically?" [Yes] [No].
Clicking either button, yes or no, does not do anything else than close the pop-up.

Any idea what can be wrong?

I've checked if there were missing dependencies, but everything required is installed.
Blueberry works and Blueman doesn't. In most cases it's the way around.

Many thanks in advance for any hint or solution,

Regards,

Marc

cschramm, 2019-07-22:
That dialog is expected starting from 2.1.beta1. Are you using a git snapshot version?

After you clicked a button, does the dialog re-appear if you start `blueman-manager` again? That probably means that there's an exception when storing your choice. It should be visible when you start `blueman-manager` in a terminal.

Defossez, 2019-07-22:
Thanks for replying so fast!
- No I'm not using a git snapshot or maybe I do. Not sure, but this is what I did.
  - I installed the repositories (deb https://cschramm.eu/blueman/debian/ python3.6 main) and then installed Blueman per **apt** command. The repository for sources is also installed.
- Yes it appears always.
  - Shutting down, start from the menu, any button in the Blueman menu. 
  - The message always appears.
- I started `blueman-manager` from within a terminal and get this message back:
```
marc@dellium:~$ blueman-manager
blueman-manager version 2.1 starting
Stale PID, overwriting
Gtk-Message: 14:29:00.158: GtkDialog mapped without a transient parent. This is discouraged.
```
The message doesn't mean much to me, I hope it tells you something :-)

Thanks again for the quick reply,

Marc

Defossez, 2019-07-22:
What I forgot in previous is:

When that text is returned in the terminal, The popup about "Shall bluetooth ...." appears. 
When I answer yes to the popup, I get this back as text in the terminal:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/Manager.py", line 112, in on_dbus_name_appeared
    lambda: Gtk.main_quit())
  File "/usr/lib/python3/dist-packages/blueman/Functions.py", line 105, in check_bluetooth_status
    config["auto-power-on"] = resp == Gtk.ResponseType.YES
  File "/usr/lib/python3/dist-packages/gi/overrides/Gio.py", line 133, in __setitem__
    self.set_value(key, GLib.Variant(type_str, value))
  File "/usr/lib/python3/dist-packages/gi/overrides/GLib.py", line 242, in __new__
    (v, rest_format, _) = creator._create(format_string, [value])
  File "/usr/lib/python3/dist-packages/gi/overrides/GLib.py", line 139, in _create
    raise NotImplementedError('cannot handle GVariant type ' + format)
NotImplementedError: cannot handle GVariant type mb
```

Installed Python3 = 3.6.8
Installed Python = 2.7.15+
  Both versions are installed by Mint 19.1

Marc

cschramm, 2019-07-22:
That's my git snapshot package, yes.

We fail to store the choice as that actually requires PyGObject 3.27.2 or later which unfortunately is not available in Ubuntu 18.04 and thus any version of Linux Mint. You'd have to get a newer version (e.g. https://packages.ubuntu.com/cosmic/python3-gi) to be able to use the git snapshots

closed 2019-07-22 by cschramm

infirit, 2019-07-23:
We should update configure.ac and DEPENDENCIES.md. I'll send a PR tonight.

cschramm, 2019-07-23:
@infirit #1078 :stuck_out_tongue_winking_eye:

infirit, 2019-07-23:
Ah, right, stil catching up on e-mails :pensive:

PR #1078 (2019-07-23): Add minimum version for PyGObject
Fixes #1077