blueman-project/blueman #1081: Issue Bluetooth don't work with Deepin rel 15.11 and rtl8821ce

bru81, 2019-07-27:
blueman: 2.04
BlueZ: 
Distribution: Deepin 15.11
Desktop environment: DDE

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

bru81, 2019-07-27:
Hi,
I've a problem with Deepin rel 15.11 and rtl8821ce bluetooth side, wifi working correctly but bluetooth doesen't work.
I don't see any adapter on bluetooth manager and I see direct firmware load error on dmesg : 

Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[    7.377579] Bluetooth: hci0: rtl: loading rtl_bt/rtl8821a_config.bin
[    7.377603] bluetooth hci0: Direct firmware load for rtl_bt/rtl8821a_config.bin failed with error -2
[    7.377605] Bluetooth: hci0: rtl: loading rtl_bt/rtl8821a_fw.bin
[    7.378605] Bluetooth: hci0: rom_version status=0 version=1
[    7.378612] Bluetooth: hci0: cfg_sz 0, total size 17428
Bluetooth: hci0: rtl: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[    7.377579] Bluetooth: hci0: rtl: loading rtl_bt/rtl8821a_config.bin
[    7.377603] bluetooth hci0: Direct firmware load for rtl_bt/rtl8821a_config.bin failed with error -2
[    7.377605] Bluetooth: hci0: rtl: loading rtl_bt/rtl8821a_fw.bin
[    7.378605] Bluetooth: hci0: rom_version status=0 version=1
[    7.378612] Bluetooth: hci0: cfg_sz 0, total size 17428

I found any workaround for others distro (as kernel patch) but for Deepin i don't found anything, someone can help me ?

infirit, 2019-07-27:
The problem is with the kernel and firmware packages for your distribution. It is best to ask on your distributions support fourum/bugzilla for help. However deepin 15.11 is based on debian stable (buster?) so any solution for it **should** work for you as well.

From what I could find this driver had problems with firmware in debian, ubuntu and arch. So very likely that deepin is affected too. In ubuntu it appears they shipped an updated linux-firmware packages which apparently fixed it.

cschramm, 2019-07-28:
You might just be missing the `firmware-realtek` package for your hardware.

LiangZhX, 2019-08-02:
Hello, I'm also trying to compile this package with Deepin, and met with some problem now.
I was wondering if I can communicate with you by email? :)

cschramm, 2019-08-02:
@LiangZhX: Feel free to open a new issue for support with compiling.

closed 2019-10-08 by stale[bot]