blueman-project/blueman #1082: Can't connect to an Android phone

robschi, 2019-07-28:
blueman: 2.1.1
BlueZ: 5.50
Distribution: Debian testing
Desktop environment: LxQT

It's not possible to connect to a Android phone with blueman with my "Intel Corporation Wireless-AC 9260" (it has Bluetooth built in).
See the attached logs.

[blueman.log](https://github.com/blueman-project/blueman/files/3439606/blueman.log)
[bluetoothd.log](https://github.com/blueman-project/blueman/files/3439607/bluetoothd.log)

cschramm, 2019-07-29:
The `kernel: Bluetooth: hci0: command ... tx timeout` messages are concerning. Your system seems to have an issue with the controller. Google yields quite a few results for the messages but most are a couple of years old and unresolved.

You could try to reload a related kernel module (probably `btintel` in your case) or reset the device in some other way. If nothing else helps, shutdown and boot (not reboot) resets the device and it might be working for a while then. You could try a different kernel / firmware version, e.g. `linux-image-5.0.0-trunk-amd64` from experimental and a manually downloaded `/lib/firmware/iwlwifi-9260-th-b0-jf-b0-43.ucode` (e.g. https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/linux-firmware.git/plain/iwlwifi-9260-th-b0-jf-b0-43.ucode).

robschi, 2019-07-29:
Yes, I was afraid that this is a firmware bug, the new firmware "iwlwifi-9260-th-b0-jf-b0-43.ucode" and kernel changed nothing.

Thank you for your help.

robschi, 2019-07-29:
Ok, nevermind, installed the new "firmware-iwlwifi" from unstable, which I guess has [this](https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/linux-firmware.git/commit/?id=7444ca40083a14682a8caa44f2c0505aa7fa43ce) fix in it, and now it works...

cschramm, 2019-07-29:
Awesome. 🙂

closed 2019-07-29 by cschramm