blueman-project/blueman #1086: systemd user service not starting

Segaja, 2019-08-06:
blueman: 2.1-1
BlueZ: 5.50-6
Distribution: archlinux
Desktop environment: i3 (and i3status for taskbar)

Currently when I try to start the `/usr/lib/systemd/user/blueman-applet.service` user service it doesn't start properly and status shows this:

```
╰─ systemctl --user status blueman-applet.service
● blueman-applet.service - Bluetooth management applet
   Loaded: loaded (/home/aschleifer/.config/systemd/user/blueman-applet.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-08-06 15:01:02 CEST; 8s ago
  Process: 30440 ExecStart=/usr/bin/blueman-applet (code=exited, status=1/FAILURE)
 Main PID: 30440 (code=exited, status=1/FAILURE)

Aug 06 15:01:02 work blueman-applet[30440]:   File "/usr/bin/blueman-applet", line 42, in <module>
Aug 06 15:01:02 work blueman-applet[30440]:     BluemanApplet()
Aug 06 15:01:02 work blueman-applet[30440]:   File "/usr/lib/python3.7/site-packages/blueman/main/Applet.py", line 17, in __init__
Aug 06 15:01:02 work blueman-applet[30440]:     setup_icon_path()
Aug 06 15:01:02 work blueman-applet[30440]:   File "/usr/lib/python3.7/site-packages/blueman/Functions.py", line 147, in setup_icon_path
Aug 06 15:01:02 work blueman-applet[30440]:     ic.prepend_search_path(ICON_PATH)
Aug 06 15:01:02 work blueman-applet[30440]: AttributeError: 'NoneType' object has no attribute 'prepend_search_path'
Aug 06 15:01:02 work systemd[3930]: blueman-applet.service: Main process exited, code=exited, status=1/FAILURE
Aug 06 15:01:02 work systemd[3930]: blueman-applet.service: Failed with result 'exit-code'.
Aug 06 15:01:02 work systemd[3930]: Failed to start Bluetooth management applet.
```

When I add the following line to the `[Service]` block in the `blueman-applet.service` file it works:

```
Environment=DISPLAY=:0.0
```

cschramm, 2019-08-06:
Do you use a desktop manager (i.e. a graphical login interface)?

Does your systemd have a `graphical-session.target`? If so, please try if adding `Requires=graphical-session.target` and / or `After=graphical-session.target` to the `[Unit]` section helps.

Segaja, 2019-08-06:
I use i3 as a window manager and SLiM as a login manager

Segaja, 2019-08-06:
My systemd has an active `graphical.target` if that is what you meant.

infirit, 2019-08-06:
``ic = Gtk.IconTheme.get_default()`` returns ``None`` which according tot the documentation should not be possible. Two things, which version of Gtk do you have and do you have a proper icon theme installed.

cschramm, 2019-08-07:
@infirit: It actually seems to be the missing `DISPLAY` variable. I do get `None` with `DISPLAY= python3 -c 'from gi.repository import Gtk; print(Gtk.IconTheme.get_default())'` as well. The question is, why is it missing... :thinking: ...or why is it set for other people... :smile:

Segaja, 2019-08-07:
I actually have gtk2 and 3 installed:

```
gtk2 2.24.32-1
gtk3 1:3.24.10-1
```

This might sound a bit stupid, but how can I see if I have a proper icon theme installed?

infirit, 2019-08-07:
@cschramm ``Gtk.IconTheme.get_default()`` will lookup the icon theme from the Gdk.Display or if that fails create a new one with Gtk.IconTheme.new. Both these functions are not supposed to return NULL/None. I can imagine if there is no DISPLAY variable the initial theme retrieval fails and creates a new one. Although this is not supposed to fail I can imagine this would be a problem if no icon theme is installed. In any case, this has nothing to do with blueman :smile:.

@Segaja for icon themes see https://wiki.archlinux.org/index.php/Icons#Icon_themes and for systemd user and the DISPLAY variable see https://wiki.archlinux.org/index.php/Systemd/User#DISPLAY_and_XAUTHORITY

cschramm, 2019-08-08:
```
DISPLAY= python3 -c 'from gi.repository import Gdk; print(Gdk.Display.get_default())'
Unable to init server: Could not connect: Connection refused
None
```

:man_shrugging:

Segaja, 2019-08-08:
@infirit I have hicolor-icon-theme installed on my arch.
The variables `DISPLAY` and `XAUTHORITY` are both set. The first says ":0.0" the second has the pass to my .Xauthority file in my home folder.

infirit, 2019-08-08:
@cschramm and this is how it is [documented](https://lazka.github.io/pgi-docs/#Gdk-3.0/classes/Screen.html#Gdk.Screen.get_default) ``Returns: a Gdk.Screen, or None if there is no default display``. But the documentation for [Gtk.IconTheme.get_default](https://lazka.github.io/pgi-docs/#Gtk-3.0/classes/IconTheme.html#Gtk.IconTheme.get_default) has ``Return type:  Gtk.IconTheme``. This may be a bug in Gtk or in the documentation. Why DISPLAY may not available for "systemd user" is discussed in the arch wiki page I linked to.

@Segaja, Sorry but there is something wrong with your setup outside of blueman's scope. The variables may be set in your I3 session but this does not automatically means it is available for "systemd user". It suggests for non standard setups to add ``systemctl --user import-environment PATH`` to ``~/.bash_profile``, try this and see if this makes a difference.

I am grumpy so the above is not personal :bowing_man:

Segaja, 2019-08-08:
Yes, running `systemctl --user import-environment` without arguments before I execute the startup of blueman-applet works. Now I just need to find a point where to put this command so that it is early enough.

Segaja, 2019-08-09:
After fixing some configuration errors on my side I know have the `DISPLAY` and `XAUTHORITY` variable in the systemd environment. My problem now is that systemd tries too soon to start the service and at that point the X server is not yet started.

I start the X server via the `startx` command automatically when I log in to tty1.

I tried to add some dependency to the `graphical.target` but so far I didn't succeed.
Do you have any suggestion about this?

infirit, 2019-08-10:
I read a bit more about this and what you are trying to do requires a lot more work. Also I am not sure we should keep our service file for blueman-applet. There is someone who provides his user unit files https://github.com/alezost/systemd-user-units.

Basically what you are now running into is that there is no --user graphical (startx) target which you need to depend on for the blueman-applet service. I has a quick look and my suggestion is to stop using systemd user and instead use i3 delivered functionality (https://i3wm.org/docs/userguide.html#_automatically_starting_applications_on_i3_startup) in combination with https://github.com/jceb/dex

It has gone way out of blueman's scope and I suggest you better take it to the i3/Arch forums if you need further assistance with starting thing in i3.

Segaja, 2019-08-10:
I talked to some people in `#archlinux` and together we found a solution.
The solutions is to create a custom target and start it in my `~/.xinitrc` file. That way I can target all my units against this new target. This also works with the `blueman-applet.service` unit.

@infirit thanks for all the help you provide. It helped me understand systemd a lot better.

closed 2019-08-10 by infirit