blueman-project/blueman #1092: blueman should take initial bluetooth mode from bluez wia dBus

h0tw4t3r, 2019-08-20:
blueman: 2.0.8-1
BlueZ: 5.50-1+b1
Distribution: ParrotOS/Debian
Desktop environment: KDE

Hi. Spent couple hours today on trying to get why blueman start's bluetooth on boot, though I have AutoEnable=false in `/etc/bluetooth/bluetooth.conf`. So far I understood that blueman has default:true for PowerManager plugin. I think it should get the AutoEnable property via dBus because it basically overrides this option, even if you have AutoEnable=false blueman with turned on PowerManager will register the device, which is actually not supposed to be.

I guess every other plugin also has the "default" values which actually have to be sent with bluez via dbus.

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-08-21:
Hi, our PowerManager's auto-power-on feature is actually older than BlueZ' AutoEnable option (added in BlueZ 5.35) and was regularly requested by users whose systems did not set up auto-enable mechanism e.g. with udev rules. As our default behavior is a constant source for confusion, blueman 2.1 asks the user for his preference, so I think your actual issue should be fixed in 2.1.

If we'd respect BlueZ' AutoEnable option, our feature would not spring into action at all as it's a supplement for cases where no auto-enable mechanism like a udev rule or AutoEnable is in place.

What D-Bus interfaces do you refer to? I'm not aware of one that exposes such BlueZ options.

h0tw4t3r, 2019-08-21:
Weii even though it is a supplement, can't we respect the AutoEnable option if so exists?

Yes, I am reffering to non-existent D-Bus interface, or well at least I would like blueman to read /etc/bluetooth/bluetooth.conf file at the startup and get the AutoEnable feature (because it's not possible to change plugin configuration on fly).

Don't get me wrong but I find it a little dumb when there is an AutoEnable feature disabled, blueman doesn't respect it and simply turns on power devices because of PowerManager applet. I've seen a lot of topics opened over the internet which ask "Why AutoEnable does not work", and I personally spent around 3 hours to get that blueman doesn't care if you have this feature or not, it puts it's own.

h0tw4t3r, 2019-08-21:
#1094 Here I tried to fix it but I am sorry I am unable to debug it properly, I will open another issue for my debugging problems on Debian.

h0tw4t3r, 2019-08-21:
btw now it doesn't work for me at all...
did `make uninstall`, `sudo apt reinstall blueman` and getting this error on running `blueman-manager`: 

```
Traceback (most recent call last):
  File "/usr/local/bin/blueman-manager", line 18, in <module>
    from blueman.main.Manager import Blueman
ModuleNotFoundError: No module named 'blueman.main.Manager'
```

cschramm, 2019-08-21:
I might be missing something but to me our feature does not make any sense if we take that setting: If `AutoEnable` is active, adapters are usually powered and there's nothing to do for us. If it is not (and there's no other auto-enable mechanism in place) - which is BlueZ' default -, our feature jumps in and powers the adapters. It does not make sense to me to disable it in that case as it's useless then. We could remove it completely instead.

Again, blueman 2.1 asks the user of he wants to enable the auto-power feature or not, so that he's aware that it exists and can be configured. That would have avoided your problem.

h0tw4t3r, 2019-08-21:
> I might be missing something but to me our feature does not make any sense if we take that setting: If `AutoEnable` is active, adapters are usually powered and there's nothing to do for us. If it is [...]

Why? Why do you in fact override driver configuration and turn on the device if it's set to be turned off?

> It does not make sense to me to disable it in that case as it's useless then. We could remove it completely instead.

What if users want to see a bluetooth applet, but don't want the bluetooth to be enabled by default? We get lots of such enchancement requests in ParrotOS, and that is exactly what I am talking about.

> Again, blueman 2.1 asks the user of he wants to enable the auto-power feature or not, so that he's aware that it exists and can be configured. That would have avoided your problem.

To me, an application which means to be a GUI to another progam should not by anyhow edit/override driver settings.

h0tw4t3r, 2019-08-21:
When you have a free minute google "how to turn off bluetooth at boot linux", and look at what users are trying to do:

https://askubuntu.com/questions/1035009/how-to-have-bluetooth-disabled-per-default
https://ubuntuforums.org/showthread.php?t=2398668
https://www.reddit.com/r/archlinux/comments/bhkxpq/bluetooth_autoenable_true_does_not_work/
https://www.reddit.com/r/Ubuntu/comments/9w3zt4/turn_off_bluetooth_on_boot/
https://askubuntu.com/questions/1076034/how-to-disable-bluetooth-on-startup-ubuntu-18-04

I am pretty sure that most of this people don't really want bluetooth.service to be turned off but want the bluetooth to be turned off.

infirit, 2019-08-21:
That escalated quickly :disappointed:.

@cschramm  explained why your PR makes no sense so i'll close it. The auto power was introduced because bluez4 always powered the adapter. When this changed in bluez5 with no option to enable it like it does now a lot of people got sad and wanted it to be powered by default. Now that bluez5 has an option @cschramm added a dialog to ask the users what they want in blueman 2.1. If you don't like it you are free to patch it in your os.

h0tw4t3r, 2019-08-21:
> bluez5 with no option to enable it like it does now 

Uhm, are you sure that bluez5 does not provide this?

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/plugins/policy.c line (854) | upstream
https://salsa.debian.org/bluetooth-team/bluez/blob/master/plugins/policy.c line (815) | debian

Taken from readme:
```
 # AutoEnable defines option to enable all controllers when they are found.
 # This includes adapters present on start as well as adapters that are plugged
 # in later on. Defaults to 'false'.
```

By default it is really commented -> that means it is disabled
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf (line 119)

For Debian, they patched bluez to have AutoEnable=true by default: https://salsa.debian.org/bluetooth-team/bluez/blob/master/debian/patches/main.conf.patch

I know that you may say "Oh, then it is a Debian problem, go and open a bug there". Not really. Since there is a way to control the initial state of adapters with blueZ, blueman should not register them by hands but has to modify the configuration file in order to change that initial state.

Having a popup dialog which reasks user for having or not having bluetooth by default is a nice idea, so user does not open the PowerManager plugin and toggles bluetooth state by himself.

Sorry if I don't understand something and putting to much emotions but that really escalated quickly. We should make the best software as possible, I will roll a new PR when I have some time.

cschramm, 2019-08-21:
> Why do you in fact override [...] configuration and turn on the device if it's set to be turned off?

The auto-power feature exists because:
- BlueZ 5 stopped auto-powering adapters which BlueZ 4 did
- BlueZ 5 < 5.35 did not have the `AutoEnable` feature at all
- BlueZ >= 5.35 does have the `AutoEnable` feature but it is disabled by default

> What if users want to see a bluetooth applet, but don't want the bluetooth to be enabled by default?

He would not need the feature then and should turn it off in the PowerManager settings or when asked for the initial setting (since 2.1).

> We get lots of such enchancement requests in ParrotOS

I'd suggest you update blueman to 2.1.1 then to give the user a choice or, if that's not possible for some reason, override the default but expect contrary requests as the result.

> To me, an application which means to be a GUI to another progam should not by anyhow edit/override [...] settings.

The thing is that BlueZ and blueman basically implemented the same feature. `AutoEnable` controls the BlueZ feature, `auto-power-on` controls its blueman counterpart. If it is enabled for both applications by default, which (unfortunately, see below) is the case in Debian with BlueZ >= 5.41 and blueman < 2.1, you need to disable both.

> I know that you may say "Oh, then it is a Debian problem, go and open a bug there".

It kinda is, yes. When @iwamatsu enabled `AutoEnable` in the `bluez` package, I should have disabled `auto-power-on` in the `blueman` package so that users get the expected result when they turn off `AutoEnable` again but never really thought about that. It's too late to file a bug for that now that blueman 2.1 is out and in Debian, sorry. :wink:

cschramm, 2019-08-21:
> Having a popup dialog which reasks user for having or not having bluetooth by default is a nice idea, so user does not open the PowerManager plugin and toggles bluetooth state by himself.

Sounds like you agree with "solved in blueman 2.1", so closing.

closed 2019-08-21 by cschramm