blueman-project/blueman #1093: Throws and Error after about 7.8 Kb

BT-RPIA, 2019-08-21:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Attempting single or multi-file transfer between RPI (Buster and Stretch OS) and between RPI and LapDesktop  and between lapDesktop and RPI stops after about 7.8kb with non-descript error, "Error occurred while sending file: name.".  The error appears on the sending RPI.  The Bluetooth Stack on the Lap/Desktop do not have this error -- the process just hangs -- the receiving RPI does not seem to know an error has happened.  NOTE:  Single or multi-file transfers between any of the Bluetooth devices I have work if the total transfer-size is less than 7.8kb.  The error message has two choices:  Restart and Cancel.  Restart does just that and if the total transfer-size it greater than about 7.8kb the error will reappear.  If Cancel is clicked, the process stops.  If nothing is done, the two devices send back and forth -- one way at about 19 B/s and the other way at about 30 B/s  of nothing known and no file ever arrives on the destination device.  It is as if the file transfer process is limited to a total of less than 7.8kb.

Installed using sudo apt-get install bluetooth bluez blueman -y on RPI B, B+, and ZERO.  I am using Buster and Stretch -- multiple SDHC 32gb cards from Samsung, Kingston and SanDisk.  I am using the provided Bluetooth Manager to setup devices and to initiate file transfer.  The exact same thing happens ((single and multiple files get sent and are received if the total transfer is less than about 7.8 kb -- if more than 7.8 kb then the error is thrown and the source and destination just aimlessly send "unknown" back and forth never delivering or sending the file(s) )) between and among any of the RPI and the Lap/Desktop.

I have reinstalled.  I have used fresh SDHC cards.  I have used SD Card Formatter, I have used balenaEtcher.  I have used Win 32 DiskImager.

cschramm, 2019-08-21:
Interesting issue :sweat_smile: 

Please start `blueman-sendto` in a terminal, providing `--loglevel debug` and provide the output for the failed transfer.

BT-RPIA, 2019-08-21:
[Image][Image]

Thank you for the prompt response. I have some pictures of the screen of the two RPI to show what I see after the error is thrown.

Right now I only have my mobile so I will respond at my earliest when I have an internet connection - may be a few days. The problem is repeatable so it can be created at my leisure.

In the meantime I will review the process to capture the requested information.  Be advised I am new at this and do not always have the knowledge to immediately inderstand what to do and how to get it done. The troubleshooting instructions looked simple enough -- I will give it a try. If I cannot do it I hope you will be open for questions.

Regards

Interesting issue 😅

BT-RPIA, 2019-08-22:
Chris

I had a moment before I had to packup.

Tried man blueman-sendto for what looked like options in your message -- nothing. If I did not do the command correctly please give the whole command -- I am a beginner at this.

Used the command blueman-sendto. I was unable to find a log file.  I just did a screen-scrape and pasted into LibreOffice Writer. The files are greater than 7.8Kb so could not Bluetooth. Copied to flash. Moved the flash to my laptop and Bluetooth to my mobile.

The first part is where I sent three small files to another RPI. I was curious what the command does. I also started the BT Manager on both RPI. The result was success as expected.

Then I started over with a large file (sand.jpg from Buster wallpaper) that would generate the unwanted though expected error. After the error appeared I clicked cancel and screen-scraped the output and pasted into the document.

Hope this will be useful to you. If I can provide more it will have to wait until I can unpack the RPI in a week or so.

If you have questions send any time. My schedule is like a roller coaster and changes frequently.

Thanks. Hope you can determine the  cause so there can be a fix.

It would be appreciated if you could hit reply so I know you received my response.

Regards

BT-RPIA, 2019-09-01:
Christoher Schramm

I re-read the troubleshooting guide multiple times. Finally determined applet NOT sendto was what I needed to use.  For a new person there is much that must be leaned by trial and mistake.  the following appeared on the screen.  I sensed it should have been sent to to butit was not so I screen scrapped and added it here.

Please advise how I am doing and if I need to do something more. Like I wrote I am new so a lot of this is over my head.

pi@PiK01:~ $ blueman-applet --loglevel debug
blueman-applet version 2.0.4 starting
There is an instance already running
pi@PiK01:~ $ killall blueman-applet
pi@PiK01:~ $ blueman-applet --loglevel debug
blueman-applet version 2.0.4 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['NetUsage', 'StatusIcon', 'PowerManager', 'GameControllerWakelock', 'AppIndicator', 'DhcpClient', 'SerialManager', 'Headset', 'StandardItems', 'KillSwitch', 'DBusService', 'NMPANSupport', 'Menu', 'PPPSupport', 'ShowConnected', 'TransferService', 'RecentConns', 'ExitItem', 'DiscvManager', 'AuthAgent', 'NMDUNSupport', 'Networking'] 
/usr/lib/python3/dist-packages/blueman/plugins/applet/AppIndicator.py:8: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import AppIndicator3 as girAppIndicator
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
Not loading PPPSupport because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
Not loading DhcpClient because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
[... 133 lines cut ...]
Describe your next action (keep empty if done): killall blueman-applet
Perform the described action
Describe your next action (keep empty if done): blueman applet --loglevel debug
Perform the described action
Describe your next action (keep empty if done): cancel bluetooth file transfer upon receipt of error
Perform the described action
Describe your next action (keep empty if done): did ctrl-c to finish
Perform the described action
Describe your next action (keep empty if done): 
Traceback (most recent call last):
  File "/usr/bin/blueman-report", line 69, in <module>
    response = urllib2.urlopen('https://api.github.com/gists', data.encode('UTF-8')).read()
  File "/usr/lib/python3.5/urllib/request.py", line 163, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.5/urllib/request.py", line 472, in open
    response = meth(req, response)
  File "/usr/lib/python3.5/urllib/request.py", line 582, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.5/urllib/request.py", line 510, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 444, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 590, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 401: Unauthorized

cschramm, 2019-09-02:
The log from `blueman-sendto` is needed. You ignored the issue template, so that I did not know if you're using a version of blueman that supports and required the `--loglevel` option. 2.0.4 does not, so just starting `blueman-sendto` from a terminal and providing the output is fine. Your previous message sounds like you already captured what's needed but it seems like you did not actually provide it anywhere yet.

BT-RPIA, 2019-09-02:
U wrote:  The log from blueman-sendto is needed. I respond: Where is this log written and what is its name?  I have no clue - please advise.

U wrote:

cschramm, 2019-09-03:
> Where is this log written

It's just the output in the terminal.

> U wrote: You ignored the issue template. I respond: I have no idea what this means.

The template you got when you created the GitHub issue, https://github.com/blueman-project/blueman/blob/master/.github/ISSUE_TEMPLATE.md. Anyway, I learned that your blueman version is 2.0.4 in the meantime.

>  i have sent two pictures and PDF files

There are image tags in https://github.com/blueman-project/blueman/issues/1093#issuecomment-523617340 but no actual files.

> Is it possible that the sudo apt-get install and update are getting an outdated version?

Depends on the definition of outdated. The versions in Debian stretch and buster are not the most recent versions of blueman, yes.

closed 2019-11-09 by stale[bot]

LRitzdorf, 2022-07-30:
If I may, I'd like to reopen this issue. I'm seeing a similar problem on Arch Linux, with Blueman 2.3.1 and Bluez 5.64. (I'm also using the i3 window manager, if that matters.)

I can send files from my paired smartphone to my laptop, but not vice versa. In my case, the transfer never actually starts, even with a text file as small as 163 bytes. I've taken the liberty of using the `blueman-sendto` command that you suggested to @BT-RPIA, and the relevant output (after discovery finishes) is below.

Also, I'd previously used Blueberry, and encountered the same issue — sending files from phone to laptop works, but not the other way around. This seems like it may be a lower-level system problem, and since I'm on Arch, it's possible that I'm missing a dependency or something.

Debug output:
```
blueman-sendto 13.48.56 INFO     Sendto:126 __init__  : Sending to [phone MAC]
blueman-sendto 13.48.56 DEBUG    DeviceSelectorWidget:63 __del__   : Deleting widget
blueman-sendto 13.48.57 INFO     Manager:58 _on_object_added: /org/bluez/obex/client/session12
blueman-sendto 13.48.57 INFO     Sendto:220 send_file : /home/me/statuses.md
blueman-sendto 13.48.57 INFO     Client:23 on_session_created: [phone MAC] [laptop MAC] opp /org/bluez/obex/client/session12
blueman-sendto 13.48.57 INFO     Manager:50 _on_object_added: /org/bluez/obex/client/session12/transfer12
blueman-sendto 13.48.57 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session12/transfer12 Status queued
blueman-sendto 13.48.57 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session12/transfer12 Name statuses.md
blueman-sendto 13.48.57 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session12/transfer12 Size 163
blueman-sendto 13.48.57 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session12/transfer12 Filename /home/me/statuses.md
blueman-sendto 13.48.57 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session12/transfer12 Transferred 0
blueman-sendto 13.48.57 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session12/transfer12 Session /org/bluez/obex/client/session12
blueman-sendto 13.48.57 INFO     ObjectPush:24 on_transfer_started: /org/bluez/obex/client/session12 /home/me/statuses.md /org/bluez/obex/client/session12/transfer12
blueman-sendto 13.48.57 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session12/transfer12 Status error
blueman-sendto 13.48.57 INFO     Manager:80 _on_transfer_completed: /org/bluez/obex/client/session12/transfer12 False
blueman-sendto 13.48.57 INFO     Manager:67 _on_object_removed: /org/bluez/obex/client/session12/transfer12
blueman-sendto 13.48.57 INFO     Manager:74 _on_object_removed: /org/bluez/obex/client/session12
blueman-sendto 13.48.57 DEBUG    Sendto:276 on_session_removed: Session removed: /org/bluez/obex/client/session12
blueman-sendto 13.48.57 INFO     Client:36 on_session_removed: /org/bluez/obex/client/session12
blueman-sendto 13.48.59 INFO     Sendto:220 send_file : /home/me/statuses.md
blueman-sendto 13.48.59 INFO     Manager:58 _on_object_added: /org/bluez/obex/client/session13
blueman-sendto 13.48.59 INFO     Sendto:220 send_file : /home/me/statuses.md
blueman-sendto 13.48.59 INFO     Client:23 on_session_created: [phone MAC] [laptop MAC] opp /org/bluez/obex/client/session13
blueman-sendto 13.48.59 INFO     Manager:50 _on_object_added: /org/bluez/obex/client/session13/transfer13
blueman-sendto 13.48.59 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session13/transfer13 Status queued
blueman-sendto 13.48.59 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session13/transfer13 Name statuses.md
blueman-sendto 13.48.59 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session13/transfer13 Size 163
blueman-sendto 13.48.59 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session13/transfer13 Filename /home/me/statuses.md
blueman-sendto 13.48.59 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session13/transfer13 Transferred 0
blueman-sendto 13.48.59 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session13/transfer13 Session /org/bluez/obex/client/session13
blueman-sendto 13.48.59 INFO     ObjectPush:24 on_transfer_started: /org/bluez/obex/client/session13 /home/me/statuses.md /org/bluez/obex/client/session13/transfer13
blueman-sendto 13.48.59 DEBUG    Transfer:44 do_g_properties_changed: /org/bluez/obex/client/session13/transfer13 Status error
blueman-sendto 13.48.59 INFO     Manager:80 _on_transfer_completed: /org/bluez/obex/client/session13/transfer13 False
blueman-sendto 13.48.59 INFO     Manager:67 _on_object_removed: /org/bluez/obex/client/session13/transfer13
blueman-sendto 13.48.59 INFO     Manager:74 _on_object_removed: /org/bluez/obex/client/session13
blueman-sendto 13.48.59 DEBUG    Sendto:276 on_session_removed: Session removed: /org/bluez/obex/client/session13
/usr/lib/python3.10/site-packages/gi/overrides/GObject.py:493: Warning: ../glib/gobject/gsignal.c:2731: instance '0x5652b3092c60' has no handler with id '2698'
  return func(*args, **kwargs)
/usr/lib/python3.10/site-packages/gi/overrides/GObject.py:493: Warning: ../glib/gobject/gsignal.c:2731: instance '0x5652b3092c60' has no handler with id '2699'
  return func(*args, **kwargs)
blueman-sendto 13.48.59 INFO     Client:36 on_session_removed: /org/bluez/obex/client/session13
```
After the initial failure message, I clicked the Retry button once before cancelling (thus the two repeated sections in the output).

LRitzdorf, 2022-07-30:
Hm, okay. After further research, it looks like the Markdown file I sent was rejected by Android, due to a MIME type filter. For anyone else encountering this in the future, see https://android.googlesource.com/platform/packages/apps/Bluetooth/+/refs/heads/master/src/com/android/bluetooth/opp/Constants.java#173 for an up-to-date list of whitelisted filetypes.