blueman-project/blueman #1095: blueman does not work on debian

h0tw4t3r, 2019-08-21:
blueman: 2.0.8-1
BlueZ: 5.50-1+b1
Distribution: parrotOS/debian
Desktop environment: KDE

blueman-mechanism process doesn't start up after building blueman from source, also reinstalling driver does not help with anything, blueman broken for me
```

┌─[✗]─[h0tw4t3r@parrot]─[~]
└──╼ $blueman-applet 
blueman-applet version 2.2 starting
There is an instance already running
┌─[h0tw4t3r@parrot]─[~]
└──╼ $blueman-manager 
blueman-manager version 2.2 starting
Stale PID, overwriting
Blueman applet needs to be running
┌─[h0tw4t3r@parrot]─[~]
└──╼ $blueman-applet 
blueman-applet version 2.2 starting
blueman-applet 14.21.36 WARNING  PluginManager:143 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-tray version 2.2 starting
Stale PID, overwriting
blueman-applet 14.22.01 ERROR    BasePlugin:96 _load     : Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms)
Traceback (most recent call last):
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
gi.repository.GLib.GError: g-dbus-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms) (20)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/BasePlugin.py", line 90, in _load
    self.on_load()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/applet/Networking.py", line 26, in on_load
    self.load_nap_settings()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/applet/Networking.py", line 47, in load_nap_settings
    m = Mechanism()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 44, in __init__
    object_path="/org/blueman/mechanism", systembus=True, *args, **kwargs)
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms)
blueman-applet 14.22.01 ERROR    PluginManager:157 __load_plugin: Failed to load Networking
Traceback (most recent call last):
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
gi.repository.GLib.GError: g-dbus-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms) (20)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/PluginManager.py", line 155, in __load_plugin
    inst._load()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/AppletPlugin.py", line 57, in _load
    super()._load()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/BasePlugin.py", line 90, in _load
    self.on_load()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/applet/Networking.py", line 26, in on_load
    self.load_nap_settings()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/applet/Networking.py", line 47, in load_nap_settings
    m = Mechanism()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 44, in __init__
    object_path="/org/blueman/mechanism", systembus=True, *args, **kwargs)
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms)
Traceback (most recent call last):
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
gi.repository.GLib.Error: g-dbus-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms) (20)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/blueman-applet", line 42, in <module>
    BluemanApplet()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/Applet.py", line 34, in __init__
    self.Plugins.load_plugin()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/PluginManager.py", line 117, in load_plugin
    self.__load_plugin(cls)
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/PluginManager.py", line 155, in __load_plugin
    inst._load()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/AppletPlugin.py", line 57, in _load
    super()._load()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/BasePlugin.py", line 90, in _load
    self.on_load()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/applet/Networking.py", line 26, in on_load
    self.load_nap_settings()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/plugins/applet/Networking.py", line 47, in load_nap_settings
    m = Mechanism()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 44, in __init__
    object_path="/org/blueman/mechanism", systembus=True, *args, **kwargs)
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms)
┌─[✗]─[h0tw4t3r@parrot]─[~]
└──╼ $Traceback (most recent call last):
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
gi.repository.GLib.Error: g-io-error-quark: Error calling StartServiceByName for org.blueman.Applet: Timeout was reached (24)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/h0tw4t3r/.local/bin/blueman-tray", line 40, in <module>
    BluemanTray()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/Tray.py", line 16, in __init__
    applet = AppletService()
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 50, in __init__
    object_path="/org/blueman/applet", *args, **kwargs)
  File "/home/h0tw4t3r/.local/lib/python3.7/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Applet: Timeout was reached

```

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-08-21:
It seems like you configured blueman with prefix `/home/h0tw4t3r/.local` which is most probably not within python's search paths when `/home/h0tw4t3r/.local/libexec/blueman-mechanism` gets executed, so it cannot find its blueman imports. `journalctl -u blueman-mechanism.service` should show that import error (assuming systemd).

You could try to add a `PYTHONPATH` to the service's environment or just install to a prefix your python interpreter will look in by default.

infirit, 2019-08-21:
My strong suggestion is not to install blueman in your home dir unless you know what you are doing (which this issue suggest you do not). For example ``blueman-mechanism`` needs its dbus service file installed in the appropriate place for it to be usable as a autostart dbus service. For most people is in ``/etc/dbus-1/system.d`` and I highly doubt you did this.

h0tw4t3r, 2019-08-21:
> My strong suggestion is not to install blueman in your home dir unless you know what you are doing (which this issue suggest you do not). For example `blueman-mechanism` needs its dbus service file [...]

I have a org.blueman.Mechanism.conf and after reinstallation I can't get `blueman-mechanism.service` to show up.

cschramm, 2019-08-21:
I think the D-Bus service configuration is in place. I would not expect the error to be a timeout otherwise. `journalctl -u blueman-mechanism.service` should show what's going wrong if you're using systemd...

h0tw4t3r, 2019-08-21:
Ok guys, I am very sorry I probably tried to install an upstream version :flushed:

However I get this problem if I compile and install it from debian source:

```
Aug 21 21:21:39 parrot org.blueman.Mechanism[826]: Traceback (most recent call last):
Aug 21 21:21:39 parrot org.blueman.Mechanism[826]:   File "/usr/local/libexec/blueman-mechanism", line 29, in <module>
Aug 21 21:21:39 parrot org.blueman.Mechanism[826]:     from _blueman import set_probe_debug
Aug 21 21:21:39 parrot org.blueman.Mechanism[826]: ImportError: No module named _blueman
Aug 21 21:21:39 parrot dbus-daemon[826]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 1
Aug 21 21:21:40 parrot dbus-daemon[826]: [system] Activating service name='org.blueman.Mechanism' requested by ':1.99' (uid=1000 pid=1792 comm="python /usr/local/bin/blueman-applet ") (using servicehelper)
```

Thanks for the help so far.

cschramm, 2019-08-21:
That's not the Debian package. It would not install anything into `/usr/local`. Neither `/usr/lib/systemd/system/blueman-mechanism.service` and `/usr/share/dbus-1/system-services/org.blueman.Mechanism.service` should refer to that binary path nor should it exist.

h0tw4t3r, 2019-08-21:
`/usr/lib/systemd/system/blueman-mechanism.service` I don't have this file at all, and I am 100% sure that it is a debian package. Here is my log: https://pastebin.com/XYhVjdkP

h0tw4t3r, 2019-08-21:
should I provide a specific prefix to the ./autogen.sh ?

h0tw4t3r, 2019-08-21:
fixed by running `./autogen.sh --prefix="/usr" --exec-prefix="/usr"`

closed 2019-08-21 by h0tw4t3r