blueman-project/blueman #1098: blueman-applet did not change state if bluetoothctl power off

RazuvaevDD, 2019-09-07:
blueman: blueman-git 2.1.alpha3.r5.g088da0c7-1
BlueZ: bluez-5.50-7
Distribution: Arch Linux
Desktop environment: Awesome WM

**Commands:**
_`$ bluetoothctl power on`
`$ blueman-applet &`
`$ bluetoothctl power off`_

**Expected:**
  blueman-applet will change its state to inactive

**Presently:**
  blueman-applet did not change its state to inactive

infirit, 2019-09-10:
Can you post the blueman-applet log as described in https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging

I can however reproduce and the relevant bit from my log is below.

```
blueman-applet 10.01.51 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Powered': False, 'Discovering': False, 'Class': 0}
blueman-applet 10.01.51 INFO     PowerManager:172 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 10.01.51 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Powered False
blueman-applet 10.01.51 INFO     DiscvManager:99 on_adapter_property_changed: Stop
blueman-applet 10.01.51 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Discovering False
blueman-applet 10.01.51 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Class 0
```

cschramm, 2019-09-15:
It looks like the PowerManager plugin just does not updates the known adapter state correctly on property change. Can you test the fix in #1102?

RazuvaevDD, 2019-09-27:
blueman-git-2.1.r32.gae2be5a7-1 in Arch repository don't work correctly.

cschramm, 2019-10-01:
What did you download exactly? To test the fix branch you need to download https://github.com/cschramm/blueman/tree/power-fix.

RazuvaevDD, 2019-10-07:
i download thus package from your link, but it's not work -_-
[image]

cschramm, 2019-10-07:
From your screenshot I'm not sure if you replaced the running applet instance. You need to exit or koll it and then run the version of `blueman-applet` you just installed.

PR #1102 (2019-10-11): Update adapter state on powered property change
Fixes #1098
commit 9720e9fd (2019-10-11): Update adapter state on powered property change

Fixes #1098
closed 2020-06-24 by RazuvaevDD