blueman-project/blueman #1100: no way to select my device priority

Askmewho, 2019-09-15:
info:
/dev/input/event0:	Logitech USB Receiver
/dev/input/event1:	Logitech USB Receiver Mouse
/dev/input/event2:	Logitech USB Receiver Consumer Control
/dev/input/event3:	Logitech USB Receiver System Control
/dev/input/event4:	ipega media gamepad controller Keyboard
/dev/input/event5:	ipega media gamepad controller Consumer Control
/dev/input/event6:	ipega media gamepad controller Mouse
/dev/input/event7:	ipega media gamepad controller

by default it will choose the event4...but Ive need event7. How to achieve that (and you may add that to the program on next releases?)

cschramm, 2019-09-15:
I do not get the question, sorry. Those event files do not compete in any way.

Askmewho, 2019-09-15:
Mmmm... okay, if you say so, i will close the issue, thanks!

closed 2019-09-15 by Askmewho