blueman-project/blueman #1107: dnsmasq not providing IP addresses for PAN

ctataryn, 2019-10-02:
blueman:
```
2.0.5-1ubuntu1 amd64
```
BlueZ:
```
5.48-0ubuntu3.1 amd64
```
Distribution:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	GalliumOS 3.0
Release:	18.04
Codename:	bionic
```

dnsmasq version:
```
$ /usr/sbin/dnsmasq --version
Dnsmasq version 2.79  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
```

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

### TL;DR
I know this is most likely a `dnsmasq` issue, but I'm curious if it is caused by [NetConf.py](https://github.com/blueman-project/blueman/blob/eb6492f1174b204863cd05d64d55ce4cca467c5c/blueman/main/NetConf.py#L55-L58) calling `dnsconf` in an improper or incomplete way and that is what's causing the issue.

For instance, I did notice this ominously named commit: "[Disable dns for DnsMasqHandler](https://github.com/blueman-project/blueman/commit/f3f958b3b70d18e5a3eaf98521119926c2d0204a#diff-bc22ed2233219bcce9f54ea185d6d0de)" in [release 2.1](https://github.com/blueman-project/blueman/blob/2.1/CHANGELOG.md#21alpha3) of `NetConf.py`, was there perhaps a known issue about DNS w.r.t. `dnsmasq`? I notice this change is still present on `master` as well and was release in v.

### Issue
Hi, I have setup the following Bluetooth PAN:
[image]

I have confirmed that checking the NAP box and hitting apply causes dnsmasq to be spawned like so:

```
/usr/sbin/dnsmasq --pid-file=/var/run/dnsmasq.pan1.pid \
--bind-interfaces --dhcp-range=10.251.112.2,10.251.112.254,60m \
--except-interface=lo --interface=pan1 \
--dhcp-option=option:router,10.251.112.1
```

And pan1 looks like this:

````
pan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.251.112.1  netmask 255.255.255.0  broadcast 10.251.112.255
        inet6 fe80::8897:3aff:fe63:27bf  prefixlen 64  scopeid 0x20<link>
        ether 8a:97:3a:63:27:bf  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 84  bytes 12199 (12.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

However when I attach a client it doesn't receive an IP address.  I have to specify the use of `dhcpd` instead of `dnsmasq` for everything to work fine.

infirit, 2019-10-02:
You should be able to kill ``dnsmasq`` started by blueman and run the command yourself, add ``-k`` to the options to keep it in the foreground. Console output may yield some hints. And you can try to adding ``--port=0`` then as well to see if that helps.

We don't actually do much other than read the dns servers from resolve.conf and start our own ``dnsmasq`` for pan1.

ctataryn, 2019-10-02:
@infirit I see, so it's not intended to provide dhcp services as well for the interface?  The command line switches passed to dnsmasq seem to indicate it should be?

cschramm, 2019-10-03:
DHCP is the only thing we use dnsmasq for. It is not intended to provide a DNS service and we thus disabled it with `--port 0` to avoid issues (see #943).

ctataryn, 2019-10-03:
@cschramm ok, we’ll it doesn’t seem to be doing the DHCP. (P.S. I often get DHCP & DNS mixed up in my head, it’s DHCP that I’m having the problem with) I’ll try @infirit’s suggestion and run `dnsmasq` with some extra settings to see why it’s not working…

infirit, 2019-10-03:
@ctataryn sorry for the confusion 😨. The name also makes it easy to get confusing, dns being in the name and also doing dhcp 😃

closed 2019-12-09 by stale[bot]

hansaplasst, 2025-04-19:
Sorry to reopen this issue but the proposed solution is more a hack than a proper fix.
I have the same problem as @ctataryn. Android is not getting an IP.

Why not start dnsmasq with a proper config file in `/etc/blueman/dnsmasq.conf`?
```
/usr/sbin/dnsmasq --pid-file=/var/run/dnsmasq.pan1.pid --conf-file=/etc/blueman/dnsmasq.conf
```
And in `/etc/blueman/dnsmasq.conf`
```
interface=pan1
except-interface=lo
listen-address=10.251.112.1
bind-interfaces
dhcp-range=10.251.112.2,10.251.112.254,60m
dhcp-option=6,8.8.8.8 # <-- Use this instead of dhcp-option=option:router,10.251.112.1 because Android doesn't work well with it.
```
This way users can modify the config to fit it to their needs.

IMHAO this solution is better than just killing dnsmasq and write a script to start dnsmasq anew.

infirit, 2025-04-19:
> Sorry to reopen this issue but the proposed solution is more a hack than a proper fix. I have the same problem as [@ctataryn](https://github.com/ctataryn). Android is not getting an IP.

Which "fix" are you referring to as no fix was provided. Only to run dnsmasq in the foreground for debugging purposes.

I just tested this to be sure and this works fine on my android phone on arch.

> Why not start dnsmasq with a proper config file in `/etc/blueman/dnsmasq.conf`?

I'm not comfortable with people providing random dnsmasq config files to blueman. And we are not adding back dns services.

How about you manage you own bridge [1], and I'll implement a way for you to provide the bridge name to blueman and we get out of your way to do whatever you want?

[1] The way this works is when registering a BlueZ `nap` NetworkServer you provide it a network bridge (name) and all traffic will be routed to that bridge. Blueman then enables ip forwarding, creates iptables rules and provides a DHCP server to give out ip addresses.

hansaplasst, 2025-04-19:
> You should be able to kill `dnsmasq` started by blueman and run the command yourself, add `-k` to the options to keep it in the foreground. Console output may yield some hints. And you can try to adding `--port=0` then as well to see if that helps.

Killing a process and then invoking another isn't a nice way to fix issues. 

> I just tested this to be sure and this works fine on my android phone on arch.

Here it's not. I'm using Ubuntu 20.04.6 LTS, bluez v5.53-0ubuntu3.9 and GrapheneOS on my ancient Android 14 phone 👴 and I only get an IPv6 which does not route through my VPN tun.

> I'm not comfortable with people providing random dnsmasq config files to blueman. And we are not adding back dns services.

What's there not to be comfortable with? If users want to invoke their own DNS services (like I do) it's just a matter of adding these options to /etc/blueman/dnsmasq.conf:
```
dhcp-option=6,[local or remote dns IP]
server=[local or remote dns IP]
```

infirit, 2025-04-19:
> What's there not to be comfortable with?

The dbus service that handles dnsmasq is run as root. I'm not going to have it accept random strings, it has to be strictly controlled by blueman.

hansaplasst, 2025-04-20:
> The dbus service that handles dnsmasq is run as root. I'm not going to have it accept random strings, it has to be strictly controlled by blueman.

I guess I'm missing the point here. Can you elaborate on this?

cschramm, 2025-04-20:
Not sure I get the problem you're trying to address. Your Android phone does not get an IPv4 address from dnsmasq if invoked with the options that blueman uses. You solve that by not setting the router option but a DNS server instead. That makes no sense, does it? Why should it get an IP v4 address then? If it does, it should be unroutable and thus also the configured DNS server is not even reachable.

You should check the DHCP traffic to investigate why your phone does not take an IPv4 address.

hansaplasst, 2025-04-23:
Here is probably not the place to explain further IMO since this issue is closed.
I'll create a new issue. Maybe we can figure out why I can't get my BT-network to work with default settings. 
Thanks for now.

ctataryn, 2025-04-23:
@hansaplasst can you drop a link to your new issue when you create it?

hansaplasst, 2025-04-23:
Sure: #2710