blueman-project/blueman #1110: Doesn't work! UwU

MikuChan03, 2019-10-11:
Hi~~
So Senpai, thewe is this issue I've been having.
I have 2 bluetooth adaptows and when I switch between them, the 3 bars that tell the signal stwength fow cuwwently connected devices disappear (；⌣̀_⌣́)

Don't wowwy though, I got it all mapped out! 🎶🎶💕

It's like this, some senpai wwote the function
set_adapter in gui/DeviceList.py, which in some cases does

```self.emit("adapter-changed", self.__adapter_path)```

twice! One such case is when simply changing an adaptor.

main/Manager.py wesponds to this signal by calling display_known_devices in gui/DeviceList.py, twice! add_device in gui/DeviceList.py is then called twice too, which then calls the sauce of the pwoblem, monitor_power_levels in gui/DeviceList.py.
monitor_power_levels checks the super fast cache that is self.monitored_devices before deciding on whether to start a timer to monitor the signal strength ot the device. This timer is passed a refewence to the Gtk.TreeRowReference that is the row with the connected device.
So monitor_power_levels runs for the first time and the timer is set up alwight, but it's one whole second long! Before that second elapses, the second invocation of display_known_devices sets up everything all over again and removes the Gtk.TreeRowReference that the first timer uses. However, when reaching monitor_power_levels for the second time, the device doesn't get handled, because it's already in the self.monitored_devices cache and no second timer is set up. So when one second finally does elapse, the first timer goes:

```if not row_ref.valid():```

So the first timer gets removed and we never get to know how stwong the signal actually is, because there is no timer that cares for it because of the cache. The super fast self.monitored_devices cache pwevents display_known_devices from being idempotent!!! You senpais can either wemove the cache or make sure to call display_known_devices only once, but as it stands, once we change between two adapters, we can't see which device is actually connected and how strong that connection is!!!!!!!!

That is all I wanted to so, I hope you can be so nice to
fixie wixie the pwoblem XD

*starts twerking*

infirit, 2019-10-11:
Thanks, see #1111 and #1112

infirit, 2019-10-13:
Merged the fixed, closing.

closed 2019-10-13 by infirit

infirit, 2019-10-27:
@KrzysztofSzewczyk you seem confused, let me help you.

@MikuChan03 identified an issue with signal bars when switching adapters. Next found the cause of the signal bars disappearing and pointed where to look. So thanks to @MikuChan03 I was able to quickly find the bug and fix it. I'm sure you see why I appreciate the contribution and we should thank @MikuChan03!

SoKamil, 2019-10-28:
what in the name of fuck

sparskakyl, 2019-12-22:
Bruh momentos numero dos

type2b, 2020-07-10:
UwU Miku chan so kawaiii *o*
*starts breakdancing*

abUwUser, 2020-07-11:
is it a joke or a owoed issue?
i just know that this was cute uwu and now i wanna hug somebody