blueman-project/blueman #1113: More than one ManagerDeviceMenu object for a device/row

infirit, 2019-10-12:
While doing the refactor discussed in the double click commit I noticed that there is more than one ManagerDeviceMenu for a device/row. Even before I noticed at least 2 before the double click event.

I don't have time right now to investigate.
```
blueman-manager 13.14.41 INFO     ManagerDeviceMenu:94 unset_op  : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f2e5c36c230 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x559f3bfcb110)>
blueman-manager 13.14.41 DEBUG    ManagerDeviceMenu:261 generate  : Beast
blueman-manager 13.14.41 INFO     ManagerDeviceMenu:94 unset_op  : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f2e5c3747d0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x559f3bfcb3f0)>
blueman-manager 13.14.41 DEBUG    ManagerDeviceMenu:261 generate  : Beast
blueman-manager 13.14.41 INFO     ManagerDeviceMenu:94 unset_op  : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f2e5c36f550 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x559f3bfcb9b0)>
blueman-manager 13.14.41 DEBUG    ManagerDeviceMenu:261 generate  : Beast
blueman-manager 13.14.41 INFO     ManagerDeviceMenu:94 unset_op  : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f2e5c37c0f0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x559f3bdc4580)>
blueman-manager 13.14.41 DEBUG    ManagerDeviceMenu:261 generate  : Beast
blueman-manager 13.14.41 INFO     ManagerDeviceMenu:94 unset_op  : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f2e5c37e0f0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x559f3bdc4b40)>
```

cschramm, 2019-10-13:
Introduced in #1108, fixed in #1114. `ManagerDeviceMenu.__init__` registers the new instance in `ManagerDeviceMenu.__instances__` from where it never gets removed as it is expected to be kept, however, in #1108 `MangerDeviceList.menu` gets reset on every double click so that we end up with unused instances that we keep regenerating.

infirit, 2019-10-13:
Ah, cool, thanks. Closing as I merged  #1114.

closed 2019-10-13 by infirit