blueman-project/blueman #1115: Page Timeout Error when pairing

Joaobranquinho, 2019-10-13:
blueman: 2.0.5
BlueZ: 5.48-0ubuntu3.1
Distribution: Ubuntu 18.04.3
Desktop environment: i3

* [ X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-10-13:
Hi João, I have no experience with that error. You should probably check the bluetoothd log, preferably with debugging enabled (see https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-bluez), as you might find a hint at what's going on there.

What BT controller are you using?

Joaobranquinho, 2019-10-13:
Here it is the logs output:
https://pastebin.com/QSrRhQW6

About the controller:
```
[NEW] Controller 54:27:1E:BE:D4:E4 envy-123 [default]
[NEW] Device C0:EE:FB:43:1E:DA OnePlus One
```

cschramm, 2019-10-13:
Seems like that status is coming directly from the firmware.

Anything in `dmesg`?

You should be able to find your controller model with `lspci`.

Joaobranquinho, 2019-10-13:
dmesg shows nothing. when trying to pair devices. However I ran `dmesg | grep Bluetooth`:
```
[   26.251944] Bluetooth: Core ver 2.22
[   26.251960] Bluetooth: HCI device and connection manager initialized
[   26.251963] Bluetooth: HCI socket layer initialized
[   26.251964] Bluetooth: L2CAP socket layer initialized
[   26.251968] Bluetooth: SCO socket layer initialized
[   27.210047] Bluetooth: hci0: BCM: chip id 63
[   27.211049] Bluetooth: hci0: BCM: features 0x07
[   27.227044] Bluetooth: hci0: envy
[   27.228038] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[   27.612068] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0a5c-21fb.hcd not found
[   43.667251] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   43.667252] Bluetooth: BNEP filters: protocol multicast
[   43.667255] Bluetooth: BNEP socket layer initialized
[  162.095982] Bluetooth: RFCOMM TTY layer initialized
[  162.095985] Bluetooth: RFCOMM socket layer initialized
[  162.095989] Bluetooth: RFCOMM ver 1.11
[ 1833.648757] Bluetooth: hci0: urb 000000002fd49f57 failed to resubmit (2)
[ 1833.885809] Bluetooth: hci0: BCM: chip id 63
[ 1833.886805] Bluetooth: hci0: BCM: features 0x07
[ 1833.902842] Bluetooth: hci0: envy
[ 1833.903806] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[ 1833.903837] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0a5c-21fb.hcd not found
[ 1842.143917] Bluetooth: hci0: urb 0000000021b70b6b failed to resubmit (2)
[ 1842.369909] Bluetooth: hci0: BCM: chip id 63
[ 1842.370905] Bluetooth: hci0: BCM: features 0x07
[ 1842.386902] Bluetooth: hci0: envy
[ 1842.387900] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[ 1842.387929] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0a5c-21fb.hcd not found
[42485.585281] Bluetooth: hci0: BCM: chip id 63
[42485.586277] Bluetooth: hci0: BCM: features 0x07
[42485.602283] Bluetooth: hci0: BCM20702A
[42485.603279] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[42485.603305] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0a5c-21fb.hcd not found
```

About the controller model:
```
08:00.0 Network controller: Broadcom Inc. and subsidiaries BCM4352 802.11ac Wireless Network Adapter (rev 03)
```

cschramm, 2019-10-13:
Seems like you're missing firmware then (the `brcm/BCM20702A1-0a5c-21fb.hcd`). https://github.com/winterheart/broadcom-bt-firmware should help.

Joaobranquinho, 2019-10-13:
Worked.
Thanks!

closed 2019-10-13 by Joaobranquinho