blueman-project/blueman #1119: Wrong output "Failed to enable bluetooth" when enable bluetooth device

parr0tr1ver, 2019-10-14:
blueman: 2.1.1
BlueZ: 5.50
Distribution: WRLinux
Desktop environment: Xfce

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

parr0tr1ver, 2019-10-14:
Why the description are missing? :(

parr0tr1ver, 2019-10-14:
@cschramm what more infomation should I provide, please?

cschramm, 2019-10-14:
The description was missing. I fixed your markup and can replicate the behavior. Thanks for the report! :slightly_smiling_face:

parr0tr1ver, 2019-10-14:
@cschramm thanks. I have been reading the source code for a while but didn't get root cause. Do you have any thought how to fix it? Thanks. :)

infirit, 2019-10-14:
@parr0tr1ver it is a timing issue. Delaying ``if not applet.GetBluetoothStatus():`` in ``check_bluetooth_status`` by a second will solve this. Now, having sleep(1) is a bit ugly so it would probably be nicer to schedule callback and have it check the status every 0.2 seconds or so. But timeout after a few seconds..

If you want to have a go at fixing this look at [Glib.timeout_add](https://lazka.github.io/pgi-docs/#GLib-2.0/functions.html#GLib.timeout_add).

Ideally we also let blueman-manager handle things without exitfunc but this requires a bit more work in ``check_bluetooth_status`` and proper testing.

zjeffer, 2023-06-10:
Almost 4 years later, This is still happening. I never have bluetooth enabled on boot, so when I start blueman-manager, I click enable bluetooth, and then it exist. This means I always have to start blueman-manager twice.

closed 2024-02-20 by cschramm