blueman-project/blueman #1130: strange import failure on master

dakkar, 2019-10-24:
blueman: HEAD at master
BlueZ: 5.51
Distribution: Gentoo
Desktop environment: none

```
$ git clone https://github.com/blueman-project/blueman.git
…
$ cd blueman/
$ ./autogen.sh
…
$ make
…
$ PYTHONPATH=$(pwd) python3.6 ./blueman/main/Applet.py 
Traceback (most recent call last):
  File "./blueman/main/Applet.py", line 3, in <module>
    import blueman.bluez as bluez
  File "/home/dakkar/src/blueman/blueman/bluez/__init__.py", line 6, in <module>
    import blueman.bluez.errors as errors
AttributeError: module 'blueman' has no attribute 'bluez'

# BUT!
$ git checkout 2.1.1
…
$ PYTHONPATH=$(pwd) python3.6 ./blueman/main/Applet.py 
$
```

I don't know enough Python to investigate further on my own. I noticed the problem because starting the applet produces that error message. It *feels* like a circular dependency somewhere.

infirit, 2019-10-28:
Running blueman uninstalled is a like black magic and also not possible for some parts. So unless you are doing development I recommend using ``blueman-9999`` which properly builds current master for you.

If you are looking to do development work let us know and I'll provide some instructions 😸.

cschramm, 2019-10-29:
Strange indeed. I'm curious what's the cause. If you have the time, could you do a git bisect to find the commit that causes the change in behavior? (a6d98931?)

dakkar, 2019-10-29:
@infirit thank you for the warning! I discovered the problem because the installed applet refused to start (with that same error message), and I did install with the 9999 Gentoo ebuild.
@cschramm good idea! I'll do that tomorrow

dakkar, 2019-10-29:
@cschramm I think you guessed right.
I wrote a little script, saved as `t`:
```bash
#!/bin/bash
PYTHONPATH=$(pwd) python3.6 ./blueman/main/Applet.py 2>/dev/null
```
then:
```
$ git checkout master
$ git bisect start HEAD 2.1 --
$ git bisect run ./t
running ./t
Bisecting: 11 revisions left to test after this (roughly 4 steps)
[9457c11cbf4b598324e311d9823c6d98c87e1c77] Add minimum version for PyGObject (#1078)
running ./t
Bisecting: 5 revisions left to test after this (roughly 3 steps)
[a6d98931859ba488c15776e82db0c707eeb311eb] Add mypy to CircleCI configuration (#1085)
running ./t
Bisecting: 2 revisions left to test after this (roughly 2 steps)
[2b2b55995241970f342e6c63182ddfe6a82a1d32] DeviceList: Remove Gtk.TreeIter from device_remove_event
running ./t
Bisecting: 0 revisions left to test after this (roughly 1 step)
[531da47b06de3617492cc5ae5530ca8b3f2e12a4] Make codebase work with basic mypy usage (#1084)
running ./t
a6d98931859ba488c15776e82db0c707eeb311eb is the first bad commit
commit a6d98931859ba488c15776e82db0c707eeb311eb
Author: Christopher Schramm <git@cschramm.eu>
Date:   Mon Aug 5 21:36:49 2019 +0200

    Add mypy to CircleCI configuration (#1085)
    
    We currently need to use --ignore-missing-imports as we do not have useful gi stubs and cannot use the following switches from --strict without changes:
    
    --disallow-subclassing-any
    --disallow-untyped-defs
    --check-untyped-defs

 .circleci/config.yml                  | 30 ++++++++++++++++++++++++++++++
 blueman/bluez/Base.py                 | 11 +++++++++--
 blueman/bluez/__init__.py             | 11 ++++++-----
 blueman/bluez/obex/__init__.py        | 14 ++++++++------
 blueman/main/DBusProxies.py           |  2 +-
 blueman/plugins/BasePlugin.py         |  5 +++--
 blueman/plugins/applet/RecentConns.py |  2 +-
 blueman/plugins/manager/Notes.py      |  3 ++-
 blueman/services/__init__.py          | 12 +++++++-----
 blueman/services/meta/__init__.py     |  6 ++++--
 10 files changed, 71 insertions(+), 25 deletions(-)
bisect run success
```

cschramm, 2019-10-29:
:thinking: I guess the changes in this file cause it somehow, also I do not see why and how (also I'm not able to reproduce your issue on my side): https://github.com/blueman-project/blueman/commit/a6d98931859ba488c15776e82db0c707eeb311eb#diff-f2637a8212efffc30aa0f552879fba1b

Some tries: 
* Does `PYTHONPATH=$(pwd) python3.6 -c 'import blueman.bluez.errors'` work fine while `PYTHONPATH=$(pwd) python3.6 -c 'import blueman.bluez.errors as errors'` fails?
* Does removing the line that sets `__all__` (in a6d9893 or later) fix the issue?
* Does `PYTHONPATH=$(pwd) python3.6 -c 'import blueman; print(blueman)'` show the correct path (i.e. `blueman/__init__.py` in `pwd`)?

dakkar, 2019-10-30:
Both commands fail the same way (this is on master 7b3923c):
```
$ PYTHONPATH=$(pwd) python3.6 -c 'import blueman.bluez.errors'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dakkar/src/blueman/blueman/bluez/__init__.py", line 6, in <module>
    import blueman.bluez.errors as errors
AttributeError: module 'blueman' has no attribute 'bluez'

$ PYTHONPATH=$(pwd) python3.6 -c 'import blueman.bluez.errors as errors'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dakkar/src/blueman/blueman/bluez/__init__.py", line 6, in <module>
    import blueman.bluez.errors as errors
AttributeError: module 'blueman' has no attribute 'bluez'
```

The import seems to happen from the right directories:
```
$ PYTHONPATH=$(pwd) python3.6 -c 'import blueman; print(blueman)'
<module 'blueman' from '/home/dakkar/src/blueman/blueman/__init__.py'>
```

Removing the assignment to `__all__` from `blueman/bluez/__init__.py` does not fix the problem, but commenting out `import blueman.bluez.errors as errors` from that file *does* make the problem go away.

I'm not sure how related this is, but after commenting out that line, this happens:
```
$ PYTHONPATH=$(pwd) python3.6 ./blueman/main/Applet.py 
Traceback (most recent call last):
  File "./blueman/main/Applet.py", line 5, in <module>
    from blueman.main.PluginManager import PersistentPluginManager
  File "/home/dakkar/src/blueman/blueman/main/PluginManager.py", line 29, in <module>
    class PluginManager(GObject.GObject, Generic[T]):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases
```

dakkar, 2019-10-30:
Also, notice how the traceback differs in these two cases:
```
$ PYTHONPATH=$(pwd) python3.6 blueman/bluez/__init__.py 
Traceback (most recent call last):
  File "blueman/bluez/__init__.py", line 2, in <module>
    from blueman.bluez.Adapter import Adapter as Adapter, AnyAdapter as AnyAdapter
  File "/home/dakkar/src/blueman/blueman/bluez/__init__.py", line 6, in <module>
    import blueman.bluez.errors as errors
AttributeError: module 'blueman' has no attribute 'bluez'

$ PYTHONPATH=$(pwd) python3.6 -c 'import blueman.bluez'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dakkar/src/blueman/blueman/bluez/__init__.py", line 6, in <module>
    import blueman.bluez.errors as errors
AttributeError: module 'blueman' has no attribute 'bluez'
``

infirit, 2019-10-30:
Happens only on python3.6 for me, python3.7 is fine, so does 3.5 but we die differently as we now use annotation only available from 3.6.

infirit, 2019-10-30:
Removing ``__all__`` makes no difference, it is with the import X as X that cause problems. Why was this needed?

cschramm, 2019-10-31:
So we have a general issue with Python 3.6 and the `import as` causes it?

I think it's actually redundant. I probably added it as a fix to missing module attributes reported by mypy but adding `__all__` also fixes the issue:

`mod1.py`
```
from blueman.bluez.Manager import Manager
```

`mod2.py`
```
from blueman.bluez.Manager import Manager as Manager
```

`mod3.py`
```
from blueman.bluez.Manager import Manager
__all__ = ["Manager"]
```

```
> mypy --ignore-missing-imports --no-implicit-reexport -c 'from mod1 import Manager'
<string>:1: error: Module 'mod1' has no attribute 'Manager'
Found 1 error in 1 file (checked 1 source file)
> mypy --ignore-missing-imports --no-implicit-reexport -c 'from mod2 import Manager'
Success: no issues found in 1 source file
> mypy --ignore-missing-imports --no-implicit-reexport -c 'from mod3 import Manager'
Success: no issues found in 1 source file
```

I'm still wondering what's causing the issue in python3.6 (seems like a python bug to me) but we should be fine with removing the `as ...` parts that I added in https://github.com/blueman-project/blueman/pull/1085/commits/cb5bde857d83bebb1878d1ddc3473acbb54b1f82

infirit, 2019-10-31:
Op do 31 okt. 2019 07:52 schreef Christopher Schramm <
notifications@github.com>:

> So we have a general issue with Python 3.6 and the import as causes it?
>

On gentoo at least, do you see the same on Debian?

> but adding __all__ also fixes the issue:
>

Perfect workaround :-).

I'm still wondering what's causing the issue in python3.6 (seems like a
> python bug to me) but we should be fine with removing the as ... parts
> that I added in cb5bde8
> <https://github.com/blueman-project/blueman/commit/cb5bde857d83bebb1878d1ddc3473acbb54b1f82>
>

It seems that way as a version before and after don't have this behaviour.

I'll see if 3.8 is available in gentoo and check it as well.

dakkar, 2019-10-31:
so, changing `from x import y as y` to `from x import y` does, indeed, remove the error message.
But now I get this:
```
$ PYTHONPATH=$(pwd) python3.6 -c 'import blueman.main.PluginManager'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dakkar/src/blueman/blueman/main/PluginManager.py", line 29, in <module>
    class PluginManager(GObject.GObject, Generic[T]):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases
```

(I didn't even know Python had metaclasses!)

infirit, 2019-10-31:
Again not a problem with 3.7. @dakkar can't you just be nice to use and use python3.7 :laughing:

dakkar, 2019-10-31:
@infirit eh ☺ at the moment I'm running blueman 2.1.1, which works with python 3.6
I have to hope that, by the time the next "stable" blueman gets tagged, Gentoo has stabilised python 3.7!
At least you can flag future blueman releases as needing python 3.7

infirit, 2019-10-31:
Nah, it actually good as I think we planned to target 3.6 for the next release. Stick below in ``/etc/portage/patches/net-wireless/blueman``, make sure it has a ``.patch`` suffix and emerge blueman-9999. You may also want to pickup https://patch-diff.githubusercontent.com/raw/blueman-project/blueman/pull/1137.patch and drop it in the same directory.

@cschramm mypy 0.730 and 0.740 are happy with below.. But I have not looked too deep so it may be wrong on other levels.

```patch
diff --git a/blueman/main/PluginManager.py b/blueman/main/PluginManager.py
index bee7fb83..4c973bf0 100644
--- a/blueman/main/PluginManager.py
+++ b/blueman/main/PluginManager.py
@@ -4,7 +4,7 @@ import os
 import logging
 import traceback
 import importlib
-from typing import Dict, Type, List, TypeVar, Generic
+from typing import Dict, List, Union
 
 from gi.repository import GObject
 
@@ -23,10 +23,7 @@ class LoadException(Exception):
     pass
 
 
-T = TypeVar('T', AppletPlugin, ManagerPlugin)
-
-
-class PluginManager(GObject.GObject, Generic[T]):
+class PluginManager(GObject.GObject):
     __gsignals__: GSignals = {
         'plugin-loaded': (GObject.SignalFlags.NO_HOOKS, None, (GObject.TYPE_STRING,)),
         'plugin-unloaded': (GObject.SignalFlags.NO_HOOKS, None, (GObject.TYPE_STRING,)),
@@ -34,8 +31,8 @@ class PluginManager(GObject.GObject, Generic[T]):
 
     def __init__(self, plugin_class, module_path, parent):
         super().__init__()
-        self.__plugins: Dict[str, T] = {}
-        self.__classes: Dict[str, Type[T]] = {}
+        self.__plugins: Dict[str, Union[AppletPlugin, ManagerPlugin]] = {}
+        self.__classes: Dict[str, Union[AppletPlugin, ManagerPlugin]] = {}
         self.__deps: Dict[str, str] = {}
         self.__cfls: Dict[str, str] = {}
         self.__loaded: List[str] = []
@@ -231,7 +228,7 @@ class PluginManager(GObject.GObject, Generic[T]):
                 args = ret
 
 
-class PersistentPluginManager(PluginManager[T]):
+class PersistentPluginManager(PluginManager):
     def __init__(self, *args):
         super().__init__(*args)
```

cschramm, 2019-11-01:
This is due to [PEP 560](https://www.python.org/dev/peps/pep-0560/) as `type(typing.Generic)` is `typing.GenericMeta` in Python 3.6 but `type` in Python 3.7.

I'm fine with the workaround. One way to make the `Generic[T]` code work with and without PEP 560 would be a version-specific metaclass for `PluginManager` like
```
try:
    from typing import GenericMeta

    class PluginManagerMeta(GObjectMeta, GenericMeta):
        pass
except ImportError:
    PluginManagerMeta = GObjectMeta
```
but I'd rather just drop Python 3.6 support to avoid that.

Unfortunately `class PluginManagerMeta(GObjectMeta, type(Generic))` does not work for Python 3.7 even though `class PluginManagerMeta(GObjectMeta, type)` (which seems equivalent to me) does.

A little off-topic: As we lack proper testing we should think about a way to catch such basic compatibility issues. Maybe a simple system that just imports each python file and checks for exceptions?

infirit, 2019-11-01:
Python 3.6 should be good to go again, let us know if you have more problems. I'll stay on 3.6 for now to avoid this :smile:

dakkar, 2019-11-15:
just built from master, commit c1f8722ccb833ea1130733e3a1fc49a46877a0b0

This happens:
```
$ blueman-applet 
blueman-applet version 2.2 starting
blueman-applet 18.50.59 ERROR    PluginManager:91 load_plugin: Unable to load plugin module NMPANSupport
Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/main/NetworkManager.py", line 6, in <module>
    gi.require_version('NM', '1.0')
  File "/usr/lib64/python3.6/site-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace NM not available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/main/PluginManager.py", line 89, in load_plugin
    importlib.import_module(self.module_path.__name__ + ".%s" % plugin)
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/applet/NMPANSupport.py", line 5, in <module>
    from blueman.main.NetworkManager import NMPANConnection
  File "/usr/lib64/python3.6/site-packages/blueman/main/NetworkManager.py", line 8, in <module>
    raise ImportError('NM python bindings not found.')
ImportError: NM python bindings not found.
blueman-applet 18.50.59 ERROR    PluginManager:91 load_plugin: Unable to load plugin module NMDUNSupport
Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/main/NetworkManager.py", line 6, in <module>
    gi.require_version('NM', '1.0')
  File "/usr/lib64/python3.6/site-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace NM not available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/main/PluginManager.py", line 89, in load_plugin
    importlib.import_module(self.module_path.__name__ + ".%s" % plugin)
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/applet/NMDUNSupport.py", line 5, in <module>
    from blueman.main.NetworkManager import NMDUNConnection
  File "/usr/lib64/python3.6/site-packages/blueman/main/NetworkManager.py", line 8, in <module>
    raise ImportError('NM python bindings not found.')
ImportError: NM python bindings not found.
blueman-applet 18.50.59 ERROR    BasePlugin:96 _load     : 'mappingproxy' object has no attribute 'setdefault'
Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/BasePlugin.py", line 90, in _load
    self.on_load()
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/applet/StatusIcon.py", line 38, in on_load
    self.query_visibility(emit=False)
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/applet/StatusIcon.py", line 70, in query_visibility
    if self.parent.Manager.get_adapters():
  File "/usr/lib64/python3.6/site-packages/blueman/bluez/Manager.py", line 70, in get_adapters
    return [Adapter(path) for path in paths]
  File "/usr/lib64/python3.6/site-packages/blueman/bluez/Manager.py", line 70, in <listcomp>
    return [Adapter(path) for path in paths]
  File "/usr/lib64/python3.6/site-packages/blueman/bluez/Base.py", line 14, in __call__
    instances: Dict[str, Dict[str, "Base"]] = cls.__dict__.setdefault("__instances__", {})
AttributeError: 'mappingproxy' object has no attribute 'setdefault'
blueman-applet 18.50.59 ERROR    PluginManager:162 __load_plugin: Failed to load StatusIcon
Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/main/PluginManager.py", line 160, in __load_plugin
    inst._load()
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/AppletPlugin.py", line 47, in _load
    super()._load()
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/BasePlugin.py", line 90, in _load
    self.on_load()
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/applet/StatusIcon.py", line 38, in on_load
    self.query_visibility(emit=False)
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/applet/StatusIcon.py", line 70, in query_visibility
    if self.parent.Manager.get_adapters():
  File "/usr/lib64/python3.6/site-packages/blueman/bluez/Manager.py", line 70, in get_adapters
    return [Adapter(path) for path in paths]
  File "/usr/lib64/python3.6/site-packages/blueman/bluez/Manager.py", line 70, in <listcomp>
    return [Adapter(path) for path in paths]
  File "/usr/lib64/python3.6/site-packages/blueman/bluez/Base.py", line 14, in __call__
    instances: Dict[str, Dict[str, "Base"]] = cls.__dict__.setdefault("__instances__", {})
AttributeError: 'mappingproxy' object has no attribute 'setdefault'
```

I don't have NetworkManager, so half those errors are expected, similar ones have always been there. The rest are new, though.

dakkar, 2019-11-15:
for comparison, with 2.1 I get:
```
$ blueman-applet 
blueman-applet version 2.1.1 starting
Stale PID, overwriting
blueman-applet 18.57.21 ERROR    PluginManager:90 load_plugin: Unable to load plugin module NMPANSupport
Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/main/NetworkManager.py", line 6, in <module>
    gi.require_version('NM', '1.0')
  File "/usr/lib64/python3.6/site-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace NM not available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/main/PluginManager.py", line 88, in load_plugin
    importlib.import_module(self.module_path.__name__ + ".%s" % plugin)
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/applet/NMPANSupport.py", line 3, in <module>
    from blueman.main.NetworkManager import NMPANConnection
  File "/usr/lib64/python3.6/site-packages/blueman/main/NetworkManager.py", line 8, in <module>
    raise ImportError('NM python bindings not found.')
ImportError: NM python bindings not found.
blueman-applet 18.57.21 ERROR    PluginManager:90 load_plugin: Unable to load plugin module NMDUNSupport
Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/main/NetworkManager.py", line 6, in <module>
    gi.require_version('NM', '1.0')
  File "/usr/lib64/python3.6/site-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace NM not available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/blueman/main/PluginManager.py", line 88, in load_plugin
    importlib.import_module(self.module_path.__name__ + ".%s" % plugin)
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib64/python3.6/site-packages/blueman/plugins/applet/NMDUNSupport.py", line 3, in <module>
    from blueman.main.NetworkManager import NMDUNConnection
  File "/usr/lib64/python3.6/site-packages/blueman/main/NetworkManager.py", line 8, in <module>
    raise ImportError('NM python bindings not found.')
ImportError: NM python bindings not found.
blueman-tray version 2.1.1 starting
```
and the program runs.

cschramm, 2019-11-17:
`AttributeError: 'mappingproxy' object has no attribute 'setdefault'` is a bug in current master that we're working on in #1143.

cschramm, 2019-12-08:
@dakkar #1143 is now in master so you should be good. :slightly_smiling_face:

dakkar, 2019-12-08:
confirmed, building from master results in a working applet!
thank you all!

closed 2019-12-08 by dakkar