blueman-project/blueman #1132: Bose headphones for for A2DP only, not HSP/HFP

mattgodbolt, 2019-10-28:
blueman: 2.1.1
BlueZ: 5.51
Distribution: Fedora 30
Desktop environment: Gnome

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2019-10-28:
We actually have nothing to do with this and it is pretty much all PulseAudio [1] and the bluetooth stack on linux BlueZ. The switching of the profile in the blueman UI just calls into pulse and you can do the same with pavucontrol (other pulseaudio mixer also I think). Speaking of pavucontrol, can you try using it to switch profile and confirm if this works or not?

You can also try ofono as described on [1] as I have never really gotten (have not tried hard though) pulseaudio to capture audio but have with it. 

Also take note of ``module-bluetooth-discover`` and experiment with changing it to "native" as mentioned on [1].

[1] https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Bluetooth/#index3h2

mattgodbolt, 2019-10-28:
Many thanks for taking the time to answer. It was a shot in the dark to know which component might be at fault, and I started at the top :) Closing, and thanks again.

mattgodbolt, 2019-10-28:
Oh, and yes, switching with `pavucontrol` has the same issues. I've googled all I can and tried all the various `audio.conf` changes, to no avail. Most folks seem to have the reverse problem...not getting A2DP :)

Thanks again!

closed 2019-10-28 by mattgodbolt

infirit, 2019-10-29:
>Oh, and yes, switching with pavucontrol has the same issues.

Thanks for confirming.

>Most folks seem to have the reverse problem...not getting A2DP :)

If I remember correctly a couple of big distros did not enable A2DP by default, I think now they do.

One thing I remembered is that sometimes missing firmware can cause strange behaviour like on my BCM20702A1 based usb dongle. Check your kernel log and look for messages that firmware loading failed. If this is the case google for the filename and put it in /lib/firmware :smile:

infirit, 2019-10-29:
Excellent! :tada:

pbabbicola, 2020-08-25:
> That was it!!! Thank you SO much. I have been upgrading firmware on the headphones, trying different bluetooth adapters, fiddling with `audio.conf` et al...ultimately using a little usb bluetooth dongle, and then finding the firmware fixed it! Hooray!

sorry for commenting on an old issue, but could you direct me to the fix? I have been fiddling a lot with the settings but cannot get it to work!