blueman-project/blueman #1136: Blueman isn't connecting to my headset like it once did. What gives!

nobleexperiment, 2019-10-31:
blueman:2.04
BlueZ: 
Distribution: ubuntu 16.04 lts
Desktop environment: yes

My Bluetooth headset is listed on recent connections (hard to screenshot the actual menu that's not visibly showin it) on the app but it's not connecting. I've done the linux lsusb, sudo service bluetooth restart blah blah everything and it's not working. My alsa stuff was even force-rebooted. I had to reinstall pulseaudio in hopes it would work again as it previously did. Now i'm stuck and exhausted all options. How can I have the blueman connect to my headset like it previously did?

link to screenshot: [https://imgur.com/dueKkHO](https://imgur.com/dueKkHO)

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-11-01:
Check `dmesg` and find the hardware ID of your controller.

nobleexperiment, 2019-11-05:
[   19.393047] audit: type=1400 audit(1572888834.649:39): apparmor="DENIED" operation="open" profile="snap.pulseaudio.pulseaudio" name="/sys/devices/virtual/dmi/id/board_vendor" pid=1576 comm="pulseaudio" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[   19.790843] audit: type=1400 audit(1572888835.045:40): apparmor="DENIED" operation="open" profile="snap.pulseaudio.pulseaudio" name="/sys/devices/virtual/dmi/id/board_vendor" pid=1576 comm="pulseaudio" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[   19.844117] audit: type=1400 audit(1572888835.101:41): apparmor="DENIED" operation="open" profile="snap.pulseaudio.pulseaudio" name="/sys/devices/virtual/dmi/id/board_vendor" pid=1576 comm="pulseaudio" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[   21.507014] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   21.507016] Bluetooth: BNEP filters: protocol multicast
[   21.507022] Bluetooth: BNEP socket layer initialized

cschramm, 2019-11-05:
So you're having some issue with your pulseaudio installation. Is your user in the `audio` group?

nobleexperiment, 2019-11-06:
nope. I type in groups audio and it spits back:~$ groups audio
groups: ‘audio’: no such user

cschramm, 2019-11-06:
That's not how to use the `groups` command.

You usually need to be in the `audio` group to be allowed to use pulseaudio.

nobleexperiment, 2019-11-08:
my user is not in the audio group. I do have pulseaudio working, but only when i use the terminal to type in Pulseaudio -D.

cschramm, 2019-11-08:
Please catch up on reading our troubleshooting page and provide the blueman-applet logs with the connection attempt.

closed 2020-01-14 by stale[bot]