blueman-project/blueman #1146: How to start blueman to the tray in i3wm?

aknyt, 2019-11-16:
blueman:
BlueZ:
Distribution: archlinux
Desktop environment: i3wm

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

aknyt, 2019-11-16:
blueman-manager is working well on my archlinux with i3wm desktop environment.

i use this command in i3wm configuration file:
`exec blueman-mananger`

blueman will autostart up on the screen directly.
but i really want blueman-manager  autostart quietly to the tray on i3wm.

I‘ve tried in LXDE, blueman can startup to tray.

checked i3wm user's wiki, no method found for this.

So, can somebody tell how to make it start to the tray on i3wm?

infirit, 2019-11-16:
blueman is split in several parts. In current stable release the main part is blueman-applet which should start blueman-tray that actually provides the icon. Look on your favourite how to start xdg autostart blueman-applet with things like dex.

closed 2019-11-16 by infirit