blueman-project/blueman #1149: Move translations to Weblate

comradekingu, 2019-11-28:
I can't accept the latest terms and agreements of Transifex, and wish to contribute to the translation of the blueman-applet.

https://weblate.org/hosting/

cschramm, 2019-11-29:
Thanks for raising this issue. Could you please elaborate a bit on the problem with Transifex as I'm not aware of it?

comradekingu, 2019-11-29:
@cschramm
It is closed source and it spies on the user. Upstream is uncooperative, and it is a very broken tool.
It also lacks features, and the ones it has are either not needed, problematic, or badly implemented.
To add insult to injury, Transifex takes ownership of translations, even prior ones.
It also does not allow for any source of monetary income for libre software projects, so that it can hold projects ransom and charge exorbitant fees once they decide to.

infirit, 2019-12-05:
There is nothing preventing weblate to add similar clauses to their tos and
sell translation.

Regardless I agree transifex sucks so if weblate provides something better
for Foss projects I'm not against.

cschramm, 2020-01-01:
Weblate looks good enough to me to consider switching, however, we have to bear in mind that we currently benefit from MATE's translation "team" on Transifex and would most probably lose most of them when we switch services. :thinking:

comradekingu, 2020-01-01:
@infirit Ownership would require sub-licensing, an interesting discussion where Weblate is as far away from Transifex as possible (?). Seeing as Weblate itself is copylefted libre software (it _is_ foss) one could always self-host if it ever could come to that.

@cschramm
That hasn't happened to any project making the switch, fortunately. This inculudes any of the projects now on their own in self-hosted instances of Weblate.
What I keep seeing is increased activity and greatly improved quality after switching. The most similar project would be [elementaryOS making the switch](https://medium.com/elementaryos/adios-transifex-weve-gone-to-weblate-59870dea06b2), (in some level of immediacy) including their Pantheon DE.

You would gain me back, and I know a few of the others from the current translation team. For the situation at large, I don't think there are any regulars not already on Hosted Weblate.

In any event, there is support for OAuth 2.0, so with just about any platform-account it is possible to log into Hosted Weblate in relative ease.

cschramm, 2020-01-11:
What I meant is that we currently benefit from the "MATE Desktop Environment team" on Transifex as blueman is set up as just a resource of the MATE project there and I doubt that many of their translators would follow us to another platform without the whole project moving.

The only viable way I could think of is using both platforms but I'm not sure how practical that is.

comradekingu, 2020-01-12:
@cschramm I was in the MATE team, and I know a lot of the best translators from there are also on Hosted Weblate. Everything on Transifex also suffers from being on Transifex, in terms of quality.
Moving is the opposite of this. It substitutes not a lot of activity for an influx of fresh eyes, new and old, with better tools.

cschramm, 2020-03-13:
I've sent a Hosting Request to Weblate but they never got back to me... :man_shrugging:

comradekingu, 2020-03-14:
I think @orangesunny can add new projects.

orangesunny, 2020-03-14:
@comradekingu is right.
I’m sorry @cschramm. I have a long queue as many free projects asked for Weblate hosting lately. I will start yours tomorrow at the latest.

orangesunny, 2020-03-15:
Up and running! https://hosted.weblate.org/projects/blueman/

Welcome aboard in our amazing community! Drop us a mail if we can help with anything else ;)

Happy translating!

orangesunny, 2020-03-15:
> There is nothing preventing weblate to add similar clauses to their tos and sell translation. Regardless I agree transifex sucks so if weblate provides something better for Foss projects I'm not against.

I promise we will never do such a thing. Weblate is and will stay libre, we love the open-source community and being part of it. That is what doing Weblate a great piece of SW.

infirit, 2020-03-15:
@orangesunny oh don't get me wrong, weblate is a great project and the hosting is much appreciated 👍 . Just pointing out the problem with any (free) hosted service, TOS may change like with Transifex. Do I expect you guys to do what Transifex did, hell no and even then I would probably be OK with it 😸 .

comradekingu, 2020-03-20:
@cschramm Thanks #1200

closed 2020-03-20 by comradekingu

cschramm, 2020-03-29:
Thanks @orangesunny. Sorry for getting back so late, I'm currently rather blocked by a long-term disaster operation.

I've just set up pushes to Weblate. I'm not 100% sure about the GitHub integration, i.e. if we will receive pull requests and in what frequency etc. :man_shrugging: We'll see.

Also I don't think new translations will properly work yet as we do not provide the pot file and Weblate thus only reads the po files. We'd have to set up some kind of build, e.g. using CircleCI, that generates the file and pushes it to Weblate... @orangesunny: Any chance to let Weblate do that internally? All our Makefiles do to generate the pot file is basically just `intltool-update --gettext-package blueman --pot` in the `po` directory.

Weblate also asks me to set a license for the translations. I don't think we had any explicit licensing information in Transifex. It probably makes sense to use GPL v3 or later to match the rest of the project. Any thoughts on that, @infirit.

infirit, 2020-03-29:
@cschramm yes, the file have a line that mentions the license is the same as the PACKAGE. PACKAGE originally was blueman but transifex messed this up and/or we have to set proper variables somewhere.

infirit, 2020-03-29:
btw I am on weblate with my github account.

orangesunny, 2020-04-05:
> Thanks @orangesunny. Sorry for getting back so late, I'm currently rather blocked by a long-term disaster operation.

No worries, I understand.

> I've just set up pushes to Weblate. I'm not 100% sure about the GitHub integration, i.e. if we will receive pull requests and in what frequency, etc. 🤷‍♂ We'll see.

You can set up the frequency in VCS configuration of the project. 24hours is the default.

> Also I don't think new translations will properly work yet as we do not provide the pot file and Weblate thus only reads the po files. We'd have to set up some kind of build, e.g. using CircleCI, [...]

Can you, please, send us an e-mail at support@weblate.org? I will check about this, thanks.

> Weblate also asks me to set a license for the translations. I don't think we had any explicit licensing information in Transifex. It probably makes sense to use GPL v3 or later to match the rest of the project. Any thoughts on that, @infirit.

@infirit is right. We even want libre projects to have translations under the same license and in the same repository as the code to be fair and clear to the translators' community. :)

cschramm, 2020-08-18:
@orangesunny: I'd like to get translations for our 2-1-stable branch. If I get it right, Weblate supports that by adding an additional component for a specific branch, however https://hosted.weblate.org/create/component/?project=460 tells me that I'm not allowed to do that although https://weblate.org/en/hosting/ says the Libre plan has the same limits as the Advanced plan and components are unlimited there (actually in all plans).

maboroshin, 2023-08-17:
Personal data is being leaked because of the Weblate migration : https://github.com/blueman-project/blueman/commit/2f88e18dce03fe0c1462eb94b66a083403de2066
GDPR defines email addresses as personal data. Weblate has already output mail addresses to Github. The feature to turn off this output was implemented at the end of 2022.

comradekingu, 2023-08-17:
@maboroshin 
It isn't leaked, it is used for the purpose of interacting with Git.
It is possible to waive provisions for even especially protected personal information,
which was done in the terms and conditions.
It would still be legal to do so, but the current solution(s) happen to be more restrictive.

The "processing of personal data for archiving purposes in the public interest, scientific or historical research purposes"
would also allow for it, arguably.

What this has to do with using Hosted Weblate now in a closed issue is anyone's guess.

If you think your user rights are better governed by TX, you are sadly mistaken.
One would be advised not to deal with them, but read their terms and services, cookie policy, etc.

orangesunny, 2023-08-17:
@comradekingu, please do not continue in this discussion. This is not a place to fight, this is a repository of a libre project.

@maboroshin, please do not call this a leak, it’s a feature. Signing commits by e-mail addresses is how Git works. To top of that:
- Every Weblate user can choose what e-mail should be used for their commits in Git. Settings are available on the Account page. Described here: https://docs.weblate.org/en/latest/user/profile.html#account
- Every project using Weblate can inform its users about specifics they care about. I think EFF is doing this well: https://github.com/EFForg/privacybadger-website#translating

If you want, @maboroshin, we can connect and chat about anything Weblate, this topic included. Our care (at) weblate dot org address seems like a good place to start for that. I would rather not fill Blueman project repo with off-topic conversation. We also use [Discussions](https://github.com/WeblateOrg/weblate/discussions).