blueman-project/blueman #1150: blueman, does not see the module

MitaiGit, 2019-12-04:
I googled 4 days unsuccessfully, help to start Bluetooth, please...
https://gist.github.com/MitaiGit/3cee53e5d0e74cd706dfa9d33b5ee509

infirit, 2019-12-05:
If the bluetoothctl command does not show any output then the adapter is
not being recognized by the kernel/bluez.

Check dmesg for errors related to this device. One very common thing is
missing firmware so make sure to look for messages like that.

MitaiGit, 2019-12-06:
you're right here's the conclusion
https://gist.github.com/MitaiGit/3cee53e5d0e74cd706dfa9d33b5ee509#gistcomment-3103246

MitaiGit, 2019-12-06:
what's next?

MitaiGit, 2019-12-06:
fucking fakes from aliexpress ((

MitaiGit, 2019-12-06:
https://pastebin.com/XdbJz569

MitaiGit, 2019-12-06:
https://linux-hardware.org/?probe=2b573d439e

MitaiGit, 2019-12-07:
https://pastebin.com/5xjqgVGL

cschramm, 2019-12-08:
What do you need help with?

Some guesses so far:
* Hardware ID 0a12:0001
* Debian buster
* No missing firmware in dmesg

You should be able to see the adapter registered to the bluetooth system then, e.g. with `bluetoothctl list` or `hcitool dev`.

MitaiGit, 2019-12-10:
https://bugzilla.kernel.org/attachment.cgi?id=107379&action=diff#a/include/net/bluetooth/hci_core.h_sec1

MitaiGit, 2019-12-10:
I so understood this error even not the first year, all business in the Chinese fakes, it is strange that in windows it started from a box without additional drivers...

MitaiGit, 2019-12-10:
are you a programmer or a student student? how to apply this patch for patching, because I'm not exactly a programmer.

MitaiGit, 2019-12-10:
https://bugzilla.kernel.org/show_bug.cgi?id=60824

MitaiGit, 2019-12-10:
is there a chance that this will be corrected in the Linux kernel?

MitaiGit, 2019-12-10:
> * No missing firmware in dmesg

can you show what firmware is in question, so I can search for information
https://pastebin.com/U7xvXex4

MitaiGit, 2019-12-10:
> * Debian buster

all completely about the system
https://linux-hardware.org/?probe=2b573d439e

cschramm, 2019-12-10:
> What do you need help with?

> You should be able to see the adapter registered to the bluetooth system then, e.g. with `bluetoothctl list` or `hcitool dev`.

MitaiGit, 2019-12-10:
`# hcitool dev
Devices:`
these commands do not work(( I gave a link to bugzilla.kernel.org

MitaiGit, 2019-12-10:
this whistle broke in kernel version 3.10, here is a patch correcting it, what is the probability that it will be poured into the kernel, I can not rebuild the kernel myself!
https://pastebin.com/mRuR99VE

MitaiGit, 2019-12-10:
это патч для старого ядра и сейчас он уже не актуален( печально что за столько лет так и не смогли починить((

cschramm, 2019-12-10:
> `# hcitool dev Devices:`
> these commands do not work((

Please elaborate on that.

> I gave a link to bugzilla.kernel.org

True. Unfortunately I cannot see how https://bugzilla.kernel.org/show_bug.cgi?id=60824 should be related to your problem...

MitaiGit, 2019-12-11:
:~$ lsusb
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) <--
Bus 001 Device 004: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 001 Device 003: ID 040b:2000 Weltrend Semiconductor 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

:~$ sudo hcitool dev
Devices:

:~$ bluetoothctl
Agent registered
[bluetooth]# list
[bluetooth]#

MitaiGit, 2019-12-11:
hciconfig
hci0:	Type: Primary  Bus: USB
	BD Address: 00:1A:7D:DA:71:11  ACL MTU: 679:9  SCO MTU: 48:16
	DOWN 
	RX bytes:574 acl:0 sco:0 events:30 errors:0
	TX bytes:368 acl:0 sco:0 commands:30 errors:0

hciconfig hci0 up
Can't init device hci0: Operation not supported (95)

MitaiGit, 2019-12-13:
what to do next? how to run a Chinese fuck-up?

MitaiGit, 2019-12-14:
пазязя((

cschramm, 2019-12-15:
Now we're getting somewhere.

That looks indeed like https://bugzilla.kernel.org/show_bug.cgi?id=60824 and you could patch your btusb module like in https://bugzilla.kernel.org/attachment.cgi?id=284525&action=diff. You can find `bcdDevice` with `lsusb` and the LMP Subversion with `hciconfig hci0 version`.

MitaiGit, 2019-12-15:
Help?

MitaiGit, 2019-12-15:
do I just need to copy that code?

cschramm, 2019-12-16:
> do I just need to copy that code?

I don't know. I gave you directions on how to find out the correct values for your device.

> I can't find where the file is btusb.c

It's a kernel driver. Sorry, but re-compiling kernel modules for your system goes far beyond the scope if this. The thread has nothing to do with blueman anyway.

closed 2019-12-16 by cschramm