blueman-project/blueman #1151: org.freedesktop.DBus.Error.NoReply

nomada-rodrigo, 2019-12-18:
blueman: 2.0.5
BlueZ: 5
Distribution: Linux Mint 19.2
Desktop environment: Mate

Description:
+- 1 to 2 minutes after connecting my bluetooth speaker Tribit Max Boom, the sound mutes. The sound preferences indicates sound is being played from the Tribit Max Boom speaker, which shows as connected in blueman too. After a while tweeking a few buttons, the speaker disappears/disconnects and blueman stops responding to mouse clicks in almost every function. I need to restart the laptop in order to reonnect the bluetooth speaker.

'blueman applet' command results:
https://gist.github.com/nomada-rodrigo/9d23adef6f6d41124b79b66fc2912676

blueman-applet --loglevel debug results:
https://gist.github.com/nomada-rodrigo/f533be484506812d4aa68fff2249f617

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

nomada-rodrigo, 2019-12-21:
Hi. I still have the issue, after a long way researching troubleshooting posts. Updated to Linux Mint 19.3 and it didn't help.
I posted it here too: https://forums.linuxmint.com/viewtopic.php?f=49&t=307394

infirit, 2019-12-21:
``org.freedesktop.DBus.Error.NoReply`` suggest we made a request to ``bluetoothd`` (BlueZ) and we never got an answer, most likely it died. It may have gotten restarted by systemd but it is very likely that version of blueman would not handle this well  Blueman 2.1 should be much better with this.

The logs you provide do not show the actual error. So start blueman-applet from a terminal, do your thing until all goes bad and then gist the output again.

ps: --loglevel does nothing for the version you are using. It is for 2.1 and later.

nomada-rodrigo, 2019-12-21:
Hi.
here it is: https://gist.github.com/nomada-rodrigo/51326bdeebff5eba0d75a80fef562aa0

I took this blueman version from the mint software manager repository. how can i install the 2.1?

cschramm, 2019-12-22:
> here it is: https://gist.github.com/nomada-rodrigo/51326bdeebff5eba0d75a80fef562aa0

There's no `NoReply` in that. Did you see it in the GUI? :thinking: 

Also check e. g. `journalctl -u bluetooth.service` for bluetoothd information.

> I took this blueman version from the mint software manager repository. how can i install the 2.1?

All available deb packages require Python 3.7, so you'd have a hard time resolving the dependencies. You could try the Python 3.6 build of master instead: https://github.com/blueman-project/blueman/wiki/Packaged-versions#debian-gnulinux It should just need a more recent python3-gi package which https://packages.debian.org/buster/python3-gi should fit. I'd only recommend following that path if you're adventurous, though. :wink:

nomada-rodrigo, 2019-12-22:
> Also check e. g. `journalctl -u bluetooth.service` for bluetoothd information.

I played a video on youtube, blueman was already on on the laptop; I just turned on the BT speaker, heard the "bleep" that tells it connects, the sound came out of the speaker for a few seconds and then stopped. That's when I did the command line and here are the results.

`rodrigo@Rodrigo-LT1:~$ journalctl -u bluetooth.service
-- Logs begin at Wed 2019-10-09 18:21:28 CEST, end at Sun 2019-12-22 15:07:02 CE
Oct 09 18:21:31 Rodrigo-LT1 systemd[1]: Starting Bluetooth service...
Oct 09 18:21:31 Rodrigo-LT1 bluetoothd[779]: Bluetooth daemon 5.48
Oct 09 18:21:31 Rodrigo-LT1 bluetoothd[779]: Starting SDP server
Oct 09 18:21:31 Rodrigo-LT1 bluetoothd[779]: Bluetooth management interface 1.14
Oct 09 18:21:31 Rodrigo-LT1 systemd[1]: Started Bluetooth service.
Oct 09 18:21:53 Rodrigo-LT1 bluetoothd[779]: Endpoint registered: sender=:1.37 p
Oct 09 18:21:53 Rodrigo-LT1 bluetoothd[779]: Endpoint registered: sender=:1.37 p
Oct 09 23:19:34 Rodrigo-LT1 bluetoothd[779]: Endpoint unregistered: sender=:1.37
Oct 09 23:19:34 Rodrigo-LT1 bluetoothd[779]: Endpoint unregistered: sender=:1.37
Oct 09 23:19:34 Rodrigo-LT1 bluetoothd[779]: Endpoint registered: sender=:1.37 p
Oct 09 23:19:34 Rodrigo-LT1 bluetoothd[779]: Endpoint registered: sender=:1.37 p
Oct 09 23:22:11 Rodrigo-LT1 bluetoothd[779]: Control: Refusing unexpected connec
Oct 09 23:22:14 Rodrigo-LT1 bluetoothd[779]: /org/bluez/hci0/dev_F4_4E_FD_8E_66_
Oct 09 23:27:58 Rodrigo-LT1 bluetoothd[779]: Suspend: Connection timed out (110)
Oct 09 23:28:08 Rodrigo-LT1 bluetoothd[779]: Suspend: Connection timed out (110)
Oct 09 23:28:10 Rodrigo-LT1 bluetoothd[779]: Abort: Connection timed out (110)
Oct 09 23:28:30 Rodrigo-LT1 bluetoothd[779]: Disconnecting failed: already disco
Oct 09 23:28:57 Rodrigo-LT1 bluetoothd[779]: connect error: Software caused conn
Oct 09 23:29:10 Rodrigo-LT1 bluetoothd[779]: Failed to set mode: Failed (0x03)
Oct 09 23:30:29 Rodrigo-LT1 bluetoothd[779]: Failed to set mode: Failed (0x03)
Oct 09 23:31:08 Rodrigo-LT1 bluetoothd[779]: Failed to set mode: Failed (0x03)
Oct 09 23:31:18 Rodrigo-LT1 bluetoothd[779]: Failed to set mode: Failed (0x03)
lines 1-23...skipping...
-- Logs begin at Wed 2019-10-09 18:21:28 CEST, end at Sun 2019-12-22 15:07:02 CET. --
Oct 09 18:21:31 Rodrigo-LT1 systemd[1]: Starting Bluetooth service...
Oct 09 18:21:31 Rodrigo-LT1 bluetoothd[779]: Bluetooth daemon 5.48
Oct 09 18:21:31 Rodrigo-LT1 bluetoothd[779]: Starting SDP server
Oct 09 18:21:31 Rodrigo-LT1 bluetoothd[779]: Bluetooth management interface 1.14 initialized
Oct 09 18:21:31 Rodrigo-LT1 systemd[1]: Started Bluetooth service.
Oct 09 18:21:53 Rodrigo-LT1 bluetoothd[779]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource
Oct 09 18:21:53 Rodrigo-LT1 bluetoothd[779]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink
Oct 09 23:19:34 Rodrigo-LT1 bluetoothd[779]: Endpoint unregistered: sender=:1.37 path=/MediaEndpoint/A2DPSource
Oct 09 23:19:34 Rodrigo-LT1 bluetoothd[779]: Endpoint unregistered: sender=:1.37 path=/MediaEndpoint/A2DPSink
Oct 09 23:19:34 Rodrigo-LT1 bluetoothd[779]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource
Oct 09 23:19:34 Rodrigo-LT1 bluetoothd[779]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink
Oct 09 23:22:11 Rodrigo-LT1 bluetoothd[779]: Control: Refusing unexpected connect
Oct 09 23:22:14 Rodrigo-LT1 bluetoothd[779]: /org/bluez/hci0/dev_F4_4E_FD_8E_66_28/fd0: fd(37) ready
Oct 09 23:27:58 Rodrigo-LT1 bluetoothd[779]: Suspend: Connection timed out (110)
Oct 09 23:28:08 Rodrigo-LT1 bluetoothd[779]: Suspend: Connection timed out (110)
Oct 09 23:28:10 Rodrigo-LT1 bluetoothd[779]: Abort: Connection timed out (110)
Oct 09 23:28:30 Rodrigo-LT1 bluetoothd[779]: Disconnecting failed: already disconnected
Oct 09 23:28:57 Rodrigo-LT1 bluetoothd[779]: connect error: Software caused connection abort (103)
Oct 09 23:29:10 Rodrigo-LT1 bluetoothd[779]: Failed to set mode: Failed (0x03)
Oct 09 23:30:29 Rodrigo-LT1 bluetoothd[779]: Failed to set mode: Failed (0x03)
Oct 09 23:31:08 Rodrigo-LT1 bluetoothd[779]: Failed to set mode: Failed (0x03)
Oct 09 23:31:18 Rodrigo-LT1 bluetoothd[779]: Failed to set mode: Failed (0x03)
Oct 09 23:31:45 Rodrigo-LT1 bluetoothd[779]: Failed to set mode: Failed (0x03)
Oct 10 03:24:54 Rodrigo-LT1 bluetoothd[779]: Terminating
Oct 10 03:24:54 Rodrigo-LT1 bluetoothd[779]: Endpoint unregistered: sender=:1.37 path=/MediaEndpoint/A2DPSource
Oct 10 03:24:54 Rodrigo-LT1 bluetoothd[779]: Endpoint unregistered: sender=:1.37 path=/MediaEndpoint/A2DPSink
Oct 10 03:24:54 Rodrigo-LT1 systemd[1]: Stopping Bluetooth service...
Oct 10 03:24:55 Rodrigo-LT1 bluetoothd[779]: Stopping SDP server
Oct 10 03:24:55 Rodrigo-LT1 bluetoothd[779]: Exit
Oct 10 03:24:55 Rodrigo-LT1 systemd[1]: Stopped Bluetooth service.
-- Reboot --
Oct 10 12:10:05 Rodrigo-LT1 systemd[1]: Starting Bluetooth service...
Oct 10 12:10:05 Rodrigo-LT1 bluetoothd[811]: Bluetooth daemon 5.48
Oct 10 12:10:05 Rodrigo-LT1 systemd[1]: Started Bluetooth service.
Oct 10 12:10:05 Rodrigo-LT1 bluetoothd[811]: Starting SDP server
Oct 10 12:10:06 Rodrigo-LT1 bluetoothd[811]: Bluetooth management interface 1.14 initialized
Oct 10 13:04:51 Rodrigo-LT1 bluetoothd[811]: Endpoint registered: sender=:1.41 path=/MediaEndpoint/A2DPSource
Oct 10 13:04:51 Rodrigo-LT1 bluetoothd[811]: Endpoint registered: sender=:1.41 path=/MediaEndpoint/A2DPSink
lines 1-39
`

nomada-rodrigo, 2019-12-24:
Followed a suggestion from sudowhammy in this post https://forums.linuxmint.com/viewtopic.php?f=49&t=307394#p1731000 , to downgrade linux firmware, but the bluetooth still crashed, and the wifi couldn't connect anymore.

Did the Gist of `journalctl -u bluetooth.service` tell you something that can give us a clue?

nomada-rodrigo, 2019-12-25:
I tried to downgrade linux firmware, after the suggestion by sudowhammy here https://forums.linuxmint.com/viewtopic.php?f=49&t=307394#p1731000

Blueman still froze.

cschramm, 2019-12-26:
I'm guessing that
```
Oct 09 23:22:11 Rodrigo-LT1 bluetoothd[779]: Control: Refusing unexpected connect
Oct 09 23:22:14 Rodrigo-LT1 bluetoothd[779]: /org/bluez/hci0/dev_F4_4E_FD_8E_66_28/fd0: fd(37) ready
```
is what you found right after the device disconnected (that was months ago...). That clearly looks like some serious firmware issue to me.

nomada-rodrigo, 2020-01-02:
> I'm guessing that
>
> ```
> Oct 09 23:22:11 Rodrigo-LT1 bluetoothd[779]: Control: Refusing unexpected connect
> Oct 09 23:22:14 Rodrigo-LT1 bluetoothd[779]: /org/bluez/hci0/dev_F4_4E_FD_8E_66_28/fd0: [...]

Is there anything I can try, besides downgrading the linux firmware which resulted in my wifi not working?

cschramm, 2020-01-04:
Upgrading to get possible fixes would be my approach (however, getting recent software into Linux Mint isn't trivial). You'll first need to find out which hardware and firmware you're using, check `lspci`. Your firmware is probably something from `linux-firmware` (I guess version 1.173.14).

nomada-rodrigo, 2020-01-09:
> Upgrading to get possible fixes would be my approach (however, getting recent software into Linux Mint isn't trivial)

Thanks cschramm. Maybe you can guide me on how to do it?

This is the result of lspci; https://gist.github.com/nomada-rodrigo/4279926c303610be68583489daa87ddf

The linux firmware is exactly 1.173.14

cschramm, 2020-01-10:
So you have a QCA9377. `ath10k_pci` should be responsible for that one. You should check `dmesg` for ath10k messages (from both the initialization and when the error takes place). [Linux Backports](https://backports.wiki.kernel.org/index.php/Main_Page) should have recent drivers and firmware.

nomada-rodrigo, 2020-01-13:
> You should check `dmesg` for ath10k messages (from both the initialization and when the error takes place)

dmesg results after bluetooth froze: https://gist.github.com/nomada-rodrigo/67b1163771f300b0ed9a3428ee1acd6d

Found many ath10k messages, but unsure to what moment they refer to. Copied some into a gist:
https://gist.github.com/nomada-rodrigo/2bbfacccb12e0d07acc37602fc8626e2

>  [Linux Backports](https://backports.wiki.kernel.org/index.php/Main_Page) should have recent drivers and firmware.

I have the system up to date. What else could I do in that sense?

infirit, 2020-01-13:
You can get the latest ath10k firmware from
https://github.com/kvalo/ath10k-firmware and see if this makes a difference.

Regardless, this is really a distribution (mint/ubuntu/debian) problem.
What you can try is boot from the latest livedvd from ubuntu and check if
it shows the same problem.

It may also be hardware failure but I agree with @cschramm it looks like
firmware is the culprit.

cschramm, 2020-01-13:
Looks like you already have the most recent firmware.

> unsupported HTC service id: 1536

yields some references around the internets. Overall the chip seems flawed / not well supported.

nomada-rodrigo, 2020-01-13:
> Looks like you already have the most recent firmware.
> 
> > unsupported HTC service id: 1536
> 
> yields some references around the internets. Overall the chip seems flawed / not well supported.

I don't know the terms you guys use very well. Is it the only left solution to try what infirit suggested?

> Regardless, this is really a distribution (mint/ubuntu/debian) problem. What you can try is boot from the latest livedvd from ubuntu and check if it shows the same problem.

closed 2020-03-20 by stale[bot]