blueman-project/blueman #1152: 蓝牙连接不成功

archlevel, 2019-12-20:
blueman:
BlueZ:
Distribution:
Desktop environment:

通过蓝牙设备工具对华为蓝牙鼠标进行设置；最后提示恭喜，成功添加设备；鼠标一直还是没能启用【在蓝牙设备工具列表上看到的华为鼠标的状态还是未知的】

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2020-01-29:
Translation,

``
Use the Bluetooth device tool to set the Huawei Bluetooth mouse. Finally, congratulations, the device was successfully added; the mouse has not been enabled. [The status of the Huawei mouse on the Bluetooth device tool list is still unknown]
``

It will be very difficult to communicate as neither @cschramm or me speak your language.

jerry1024-jif, 2020-02-29:
I have the same problem as above; it works on deepin OS, but cannot pair on ubuntu or fedora
[bluetoothctl messages.txt](https://github.com/blueman-project/blueman/files/4270038/bluetoothctl.messages.txt)

cschramm, 2020-02-29:
@jerry1024-jif: That's out of blueman 's scope. For debugging I propose to look at bluetoothd's log output, possibly running it with debugging ouput enabled.

closed 2020-05-06 by stale[bot]