blueman-project/blueman #1154: Unstable sometimes crash on change audio profile A2DP -> HSP

dansilver314, 2019-12-30:
blueman: 2.0.5
BlueZ: 5.48
Distribution: 4.15.0-54-generic / Linux Mint 19.2 Tina
Desktop environment: 
RELEASE=19.2
CODENAME=tina
EDITION="MATE"
DESCRIPTION="Linux Mint 19.2 Tina"
DESKTOP=Gnome
TOOLKIT=GTK

* [X YES ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

dansilver314, 2019-12-30:
Not ever, but sometimes crash happens and only way to fix that is restarting my laptop(restarting service doesn't help).

To improve: i work lot of hours every day doing lot of calls via google hangout and comeback to listen youtube music.
I need to change almost 15 times every day (go and back = 30 times) from A2DP ->HSP and vice versa to continue listen music.
If is possible in the future to add some autodetection for that instead to change 30 times every day, will be great.
Thank you!

cschramm, 2020-01-01:
Please elaborate on the term crash.

> If is possible in the future to add some autodetection for that instead to change 30 times every day, will be great.

Your version of pulseaudio does have some support for auto switching, see e.g. https://www.freedesktop.org/wiki/Software/PulseAudio/Notes/11.0/#optiontoautomaticallyswitchbluetoothprofiletohspmoreoften

closed 2020-03-08 by stale[bot]