blueman-project/blueman #1158: Newer versions for 18.0.4

SteelChickens, 2020-01-04:
Any way to use a 2.1 version on linux mint. All of the 18.0.4 packages are stuck at 2.0.5

I tried to install the ubuntu 18.10 version and have the correct python but some windows like "device" do not open.

cschramm, 2020-01-04:
> Any way to use a 2.1 version on linux mint.

The dependencies should be fine except for the PyGObject version which would cause a specific issue, see #1077. Ubuntu 19.04 was the first release with a sufficient one.

> I tried to install the ubuntu 18.10 version

Ubuntu 18.10 had blueman 2.0.5 as well.

> and have the correct python but some windows like "device" do not open.

Please elaborate.

fsevenm, 2020-02-05:
I guess it is less possible to install or build the newer version of blueman on 18.04, because of its dependencies. I had tried to download the source then built it, but it failed because it used some newest python libraries those 18.04 distributions don't have.
I insisted to fix dependencies by upgrading some needed dependencies, but it resulted many of my apt packages and apps broken. So, I removed my intention to upgrade it. I fixed my broken pkgs and keep using the blueman 2.0.5 which is supported by 18.04 so far.

infirit, 2020-02-06:
The commit that is responsible for the pygobject version requirement is 19c59502cea9673d4b944a6d8950ee5dbfae05fe. If people want they could revert this and apply this to their own build. 

Totally unsupported but here is a patch against 2.1 stable branch, https://gist.github.com/infirit/50d8d0812da14cb8506279c04e3f73b2