blueman-project/blueman #1159: 2.1.1: Can't pair with feitian U2F BT token

Thermi, 2020-01-09:
blueman: 2.1.1
BlueZ: 5.52
Distribution: Arch Linux
Desktop environment: XFCE4

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

It goes as far as prompting for the key, but then does nothing when hitting okay.
When running blueman from the terminal, I get the following output:

```
(blueman-tray:79548): Gdk-CRITICAL **: 16:36:17.940: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
blueman-manager version 2.1.1 starting
Stale PID, overwriting
blueman-manager 16.36.23 WARNING  DeviceList:193 monitor_power_levels: Failed to get power levels, probably a LE device.
blueman-manager 16.36.33 WARNING  DeviceList:193 monitor_power_levels: Failed to get power levels, probably a LE device.
/usr/lib/python3.8/site-packages/gi/overrides/Gtk.py:1630: Warning: g_value_get_int: assertion 'G_VALUE_HOLDS_INT (value)' failed
  return _Gtk_main(*args, **kwargs)
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 130, in _create
    iter(value)
TypeError: 'int' object is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/blueman/main/applet/BluezAgent.py", line 172, in passkey_dialog_cb
    ret = GLib.Variant('(u)', int(ret))
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 189, in __new__
    v = creator._create(format_string, value)
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 132, in _create
    raise TypeError("Could not create array, tuple or dictionary entry from non iterable value %s %s" %
TypeError: Could not create array, tuple or dictionary entry from non iterable value (u) 401704
blueman-manager 16.37.04 ERROR    Manager:201 error_handler:  Authentication Canceled
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/blueman/bluez/Base.py", line 81, in callback
    value = proxy.call_finish(result).unpack()
gi.repository.GLib.GError: g-io-error-quark: GDBus.Error:org.bluez.Error.AuthenticationCanceled: Authentication Canceled (36) 
```

cschramm, 2020-01-10:
Please check the bluetoothd log to get more insights on the cancelation, possibly with debugging enabled, see https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-bluez.

Thermi, 2020-01-10:
The problem is actually with the Python exception.

cschramm, 2020-01-10:
:sleepy: Your probably absolutely right. Could you try the following patch:

```
diff --git a/blueman/main/applet/BluezAgent.py b/blueman/main/applet/BluezAgent.py
index c28825bd..68e1648f 100644
--- a/blueman/main/applet/BluezAgent.py
+++ b/blueman/main/applet/BluezAgent.py
@@ -169,7 +169,7 @@ class BluezAgent(DbusService):
             if response_id == Gtk.ResponseType.ACCEPT:
                 ret = pin_entry.get_text()
                 if is_numeric:
-                    ret = GLib.Variant('(u)', int(ret))
+                    ret = GLib.Variant('u', int(ret))
                 ok(ret)
             else:
                 err(BluezErrorRejected("Rejected"))
```

Just apply it to your /usr/lib/python3.8/site-packages/blueman/main/applet/BluezAgent.py either with `patch(1)` or by manually editing the file. Restart blueman-applet afterward to test it.

Thermi, 2020-01-10:
No luck just yet:
```
blueman-applet version 2.1.1 starting
blueman-tray version 2.1.1 starting
Terminating blueman-tray
blueman-tray version 2.1.1 starting
Stale PID, overwriting

(blueman-tray:42275): Gdk-CRITICAL **: 10:55:14.916: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
blueman-manager version 2.1.1 starting
Stale PID, overwriting
blueman-manager 10.55.33 WARNING  DeviceList:193 monitor_power_levels: Failed to get power levels, probably a LE device.
/usr/lib/python3.8/site-packages/gi/overrides/Gtk.py:1630: Warning: g_value_get_int: assertion 'G_VALUE_HOLDS_INT (value)' failed
  return _Gtk_main(*args, **kwargs)
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/blueman/main/applet/BluezAgent.py", line 173, in passkey_dialog_cb
    ok(ret)
  File "/usr/lib/python3.8/site-packages/blueman/main/DbusService.py", line 116, in ok
    invocation.return_value(self._prepare_arguments(result_signature, result))
  File "/usr/lib/python3.8/site-packages/blueman/main/DbusService.py", line 145, in _prepare_arguments
    return GLib.Variant("(%s)" % signature, args) if signature else None
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 189, in __new__
    v = creator._create(format_string, value)
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 150, in _create
    builder.add_value(self._create(dup, i))
  File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 118, in _create
    return self._LEAF_CONSTRUCTORS[format](value)
TypeError: Must be number, not Variant
```

cschramm, 2020-01-10:
Then that line should probably just be `ret = int(ret)`.

Thermi, 2020-01-10:
That worked.

cschramm, 2020-01-10:
Awesome, thanks for the verification. I'll prepare a merge request for this. 🙂

infirit, 2020-01-28:
Fix released

commit 4858fa6d (2020-01-11): Fix pincode entry response

Closes #1159
PR #1160 (2020-01-13): Fix pincode entry response
Closes #1159
commit 39220e02 (2020-01-13): Fix pincode entry response

Closes #1159
closed 2020-01-28 by infirit