blueman-project/blueman #1163: Bluetooth is off but "1 Active connection" is showed

kuraga, 2020-01-13:
blueman: 2.1.1
BlueZ:
Distribution: Calculate Linux 20
Desktop environment: XFCE 4.14

[image]

infirit, 2020-01-28:
Yeah we never remove the second line in the tooltip when turning the power off. Below should do the trick.

```diff
diff --git a/blueman/plugins/applet/ShowConnected.py b/blueman/plugins/applet/ShowConnected.py
index 0fe54383..083ee441 100644
--- a/blueman/plugins/applet/ShowConnected.py
+++ b/blueman/plugins/applet/ShowConnected.py
@@ -50,6 +50,7 @@ class ShowConnected(AppletPlugin):
                                                                      "<b>%d Active Connections</b>",
                                                                      self.num_connections) % self.num_connections)
         else:
+            self.parent.Plugins.StatusIcon.set_text_line(1, None)
             # bluetooth should be always enabled if powermanager is not loaded
             status = True
             if 'PowerManager' in self.parent.Plugins.get_loaded():
```

kuraga, 2020-01-28:
I think we have to deferment/set to zero number of connections?

kuraga, 2020-01-28:
I think we have to set to zero/decrement number of connections?

infirit, 2020-01-28:
> I think we have to set to zero/decrement number of connections?

This is already done [here](https://github.com/blueman-project/blueman/blob/a456b17496c9c751983a7252c9d0aadda253bc51/blueman/plugins/applet/ShowConnected.py#L70). The only thing missing was actually updating the tooltip :smile:

infirit, 2020-01-28:
See #1170

infirit, 2020-05-16:
#1170 was merged so closing.

closed 2020-05-16 by infirit