blueman-project/blueman #1164: Bluetooth cannot search for a device or add a device

iobami, 2020-01-17:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

mlk71075, 2020-01-25:
Confirm. Ubuntu Focal, blueman-manager 2.1.1, scanning can't see any new device.
ii  blueman                              2.1.1-1                           amd64        Graphical bluetooth manager
ii  libbluetooth3:amd64                  5.52-0ubuntu2                     amd64        Library to use the BlueZ Linux Bluetooth stack
ii  libgnome-bluetooth13:amd64           3.34.0-1                          amd64        GNOME Bluetooth tools - support library
ii  pulseaudio-module-bluetooth          1:13.0-3ubuntu1                   amd64        Bluetooth module for PulseAudio sound server

cschramm, 2020-01-26:
@mlk71075: Are you sure that you do have the same issue as @iobami? You seem to be able to search for devices, @iobami does not.

Check https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman regarding a blueman-applet debug log of the scan. Also check if devices show up in bluetoothctl. If yes, this is not a blueman issue but something more fundamental in your in Bluetooth stack.

mlk71075, 2020-01-26:
I am sure. I did both, no device appeared.
However on another machine, I could successfully scan for devices. Same
Ubuntu version, both updated before I wrote the comment.

franzever, 2020-03-30:
tengo el mismo problema.
soy nuevo en el mundo de linux ubuntu acabo de instalarlo y el dispositivo de bluetooth no logra identificar a otros cuando los busca. 
busque por la web configuraciones posibles soluciones con blueman y bluez pero hasta ahora nada, no detecta dispositivos moviles, tabletas o equipos de audio.
alguien podria ayudarme..

closed 2020-06-05 by stale[bot]