blueman-project/blueman #1166: the added icon cannot be clicked

cws999, 2020-01-22:
blueman:2.0.4
BlueZ:5.43-2
Distribution:devuan
Desktop environment:xfce

i run blueman-manager,On the toolbar,Search and delete icons can be clicked,But the added icon cannot be clicked.So i can't add my bluetooth keyboard.
Here is  terminal  prompt

> 
> *****@*******:~$ blueman-manager
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:154:27: The style property GtkButton:image-spacing is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:155:26: The style property GtkButton:inner-border is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:584:27: The style property GtkButton:image-spacing is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:585:28: The style property GtkWidget:interior-focus is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:586:28: The style property GtkButton:default-border is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:587:26: The style property GtkButton:inner-border is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:1344:25: The style property GtkRange:slider-width is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:1345:25: The style property GtkRange:stepper-size is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:1346:34: The style property GtkScrollbar:min-slider-length is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:1347:28: The style property GtkRange:stepper-spacing is deprecated and shouldn't be used anymore. It will be removed in a future version
> 
> (blueman-manager:4302): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:1348:34: The style property GtkRange:trough-under-steppers is deprecated and shouldn't be used anymore. It will be removed in a future version
> _________
> Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
> ['Services', 'PulseAudioProfile'] 
> _________
> __load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
> loading <class 'blueman.plugins.manager.Services.Services'> 
> _________
> __load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
> loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
> _________
> pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
> 1 
> blueman-manager version 2.0.4 starting
> _________
> pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
> 2 
> _________
> on_bluez_name_owner_changed (/usr/bin/blueman-manager:96)
> org.bluez owner changed to  :1.4 
> _________
> get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
> Detected BlueZ 5 
> _________
> SetAdapter (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:271)
>  
> _________
> on_adapter_changed (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerToolbar.py:83)
> toolbar adapter /org/bluez/hci0 
> _________
> pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
> 3 
> _________
> pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
> 4 
> _________
> on_pa_ready (/usr/lib/python3/dist-packages/blueman/plugins/manager/PulseAudioProfile.py:29)
> connected 
> _________
> <lambda> (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:353)
> 1

infirit, 2020-01-22:
The ui may be a bit confusing depending on the icon theme. The plus icon means pair not add a device. What you should do is hit search, select the device you want to pair (add) and then click on the plus icon.

cws999, 2020-01-24:
> The ui may be a bit confusing depending on the icon theme. The plus icon means pair not add a device. What you should do is hit search, select the device you want to pair (add) and then click on the plus icon.

Devices are searchable，But cannot add

cschramm, 2020-01-26:
If the pairing button in the toolbar is not sensitive (usually grayed out, depending on the UI), no device is selected or the selected device is already paired.

cws999, 2020-01-31:
> If the pairing button in the toolbar is not sensitive (usually grayed out, depending on the UI), no device is selected or the selected device is already paired.

Device selected
The selected device is not paired

cschramm, 2020-02-01:
Please provide a screenshot.

cws999, 2020-02-02:
> Please provide a screenshot.

[https://send.firefox.com/download/5358fdb236880bbe/#9EHZHfZSId7pzWfb5ZnWOQ](url)

closed 2020-04-09 by stale[bot]