blueman-project/blueman #1167: Configure default connection mode paired devices

joapuiib, 2020-01-22:
I've paired a Sony WH-1000XM2 to my laptop using blueman. 
Every time I turn on the headphones, it connects automatically to my computer as __Headset__. They need to be connected as __Audio Sink__ in order to work properly, so I have to open the Blueman Manager, disconnect them and reconnect them manually as Audio Sink.

Is it possible to configure it so my headphones connect as __Audio Sink__ by defualt?

infirit, 2020-01-22:
Short answer, not in blueman but maybe and sort of in puselaudio.

Long answer, blueman uses the bluez dbus api to control pairing, connecting etc. For audio we actually have no control **on connect** what happens as audio is part of what is called "auto connect" profiles which also includes HID (pointer, keyboard etc). We do allow you to change the profile but this is actually a blueman plugin that talks to pulseaudio.

So BlueZ does not handle audio itself (it did in the past). Instead it provides interfaces for other programs to register themselves that they will handle audio. On most distributions this is done by pulseuadio but since a year or so bluealsa is also more commonly used (mostly on embedded and raspberry like devices).

Connect your device and run

``
pacmd list-cards
``

It will give you a bunch of detailed information on your audio cards. It also show which profile for bluetooth has priority which may explain why it connects headset by default. Find the name of your card which is for my phone is ``bluez_card.F4_60_E2_C4_96_3F``. Also figure out the exact profile name which you'll need next.

Then set the profile with

``
pacmd set-card-profile bluez_card.F4_60_E2_C4_96_3F <profile name>
``

If I'm now mistaken you can also add this in ``/etc/pulse/default.pa`` but then leave of ``pacmd``.

joapuiib, 2020-01-22:
Thanks for the long answer, that was what I was looking for.

closed 2020-01-22 by joapuiib