blueman-project/blueman #1168: Refresh Device List does not work

LinJLK, 2020-01-27:
blueman: 2.0.5
BlueZ:5.48-0ubuntu3.2
Kernel: 4.15.0-70-generic x86_64 bits: 64 
Desktop: Cinnamon 4.0.10 Distro: Linux Mint 19.1 Tessa 

Where could be problem? When i try to refresh the device list / search for new devices nearby it only shows my favorit devices and some old devices from the first start of Blueman. Try to pair an old bluetooth keyboard and some new sony bluetooth headset both didn't show up. 

Thanx

cschramm, 2020-01-27:
Hi @LinJLK, please run `bluetoothctl` in parallel during the scan to see if devices show up their. If they do not, it's a more fundamental issue in your Bluetooth stack and not related to blueman.

LinJLK, 2020-01-27:
Hi @cschramm, thanks for your quick reply. I got it to work via bluetoothctl. First there was this error message "Failed to start discovery: org.bluez.Error.InProgress" i think because it was already started but then after another try it worked.

I also tried `hcitool scan` to get my old apple keyboard to work. Not easy to pair though but i am working on it.

Thank u best regs

cschramm, 2020-01-27:
So devices *do* show up in `bluetoothctl` but *not* in `blueman-manager` during a discovery (no matter if you start in `blueman-manager` or `bluetoothctl`)? Can we get a `blueman-applet` log of that? See https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman.

LinJLK, 2020-01-27:
No. I think i just didn't have enough patience. It looks like when I started the search in blueman in the  nothing new showed up at first.
 So i started this issue. Now its working fine. Maybe because i restarted it properly and had a look into bluetoothctl. Just guessing. Here is the log <script src="https://gist.github.com/LinJLK/cfd1509b40a1e60a6d1eb9c7ca5e0dee.js"></script>

closed 2020-01-27 by cschramm