blueman-project/blueman #1172: Bluetooth adaptor presenting misbehave

BrunoXL, 2020-02-02:
blueman: 2.0.8
BlueZ: 5
Distribution: 19.10 (Eonian)
Desktop environment: Gnome

* [X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I'm trying to connect a device using a bluetooth dongle. 
After plugged it on my computer, Blueman seems to present misbehave: it applets image seems to be `on`, but the menu when clicked indicates it is `off`. 
The opposite happens as well (applet is `off` and menu content is `on`).

[image]
[image]

[Debug logs](https://gist.github.com/BrunoXL/dca3de756def15f011ac28b8a810d4ca.js)

BrunoXL, 2020-02-10:
As anyone could find time to answer me I find a solution [here](https://askubuntu.com/questions/1205962/bluetooth-dongle-problem/1208994?noredirect=1#comment2030357_1208994)

closed 2020-02-10 by BrunoXL

cschramm, 2020-02-10:
Hi Bruno, looks like an issue with your icon theme. Blueman an icon with blue background and a similar with the red cross for when Bluetooth is turned of. Your theme obviously overrides the default icon but not the disabled variant. I cannot relate this to the comment you linked to.