blueman-project/blueman #1173: dhclient dbus error when connecting to PAN on a cellhpone

baybal, 2020-02-05:
blueman:2.1.2
BlueZ:5.52
Distribution:Gentoo
Desktop environment:Xfce

* [*] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

https://pastebin.com/bV9mjDYV

When I try to connect to PAN server on my cellphone, I receive "Argument 0 does not allow None as a value" in File "/usr/lib64/python3.6/site-packages/blueman/plugins/mechanism/Network.py", line 30, in dh_connected

It used to work, but I suspect the recent dhcpcd update could be to blame

infirit, 2020-02-07:
Can you grep the emerge.log for dhcpcd ``grep dhcpcd /var/log/emerge.log`` for the version you had before. Does downgrading to that version make it work? And which version does not work, the one currently installed ``equery l net-misc/dhcpcd``.

infirit, 2020-02-07:
Also, can you post the output of ``/usr/libexec/blueman-mechanism -d -s --loglevel debug`` when you try to connect.

Looking at this code there are several problems outside of the TypeError.

In the blueman applet plugin we have a timeout on the dbusproxy is 120 ms which is, obviously, way to short. Also we appear to never kill the client when we disconnect.

For the timeout see e52e05c01ffa229963b97361870c50b771297a00 on PR #1177. I'll update that PR as I go fixing this.

baybal, 2020-02-13:
> version you had before

7.2.3

> which version does not work, the one currently installed

8.1.5

> can you post the output of /usr/libexec/blueman-mechanism -d -s --loglevel debug when you try to connect.

(blueman-mechanism:9111): dbind-WARNING **: 17:41:36.299: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
blueman-mechanism 17.41.36 INFO     blueman-mechanism:69 <module>  : Enabled verbose output
blueman-mechanism 17.41.36 INFO     blueman-mechanism:78 <module>  : Starting blueman-mechanism
blueman-mechanism 17.41.36 INFO     blueman-mechanism:148 __init__  : loading Ppp
blueman-mechanism 17.41.36 INFO     blueman-mechanism:148 __init__  : loading Network
blueman-mechanism 17.41.36 INFO     blueman-mechanism:148 __init__  : loading Rfcomm
blueman-mechanism 17.41.36 INFO     blueman-mechanism:148 __init__  : loading RfKill

baybal, 2020-02-13:
I just downgraded to 7.2.3. Still not working.

Something else must've been broken

cschramm, 2020-02-15:
That log does not look like you tried to connect.

Tracing down how this could happen:
The exception says `ip` is `None` in https://github.com/blueman-project/blueman/blob/2.1.2/blueman/plugins/mechanism/Network.py#L30.
`get_net_address(self._interface)` in https://github.com/blueman-project/blueman/blob/2.1.2/blueman/main/DhcpClient.py#L62 must have returned `None`. That means either `self._interface` was `None` which does not seem possible to me or an error happend in `get_net_address`, probably down in the C function... :thinking: 

Those things changed a lot recently. Could you try master?

infirit, 2020-02-16:
ok(ip) is in response to the ``connected`` signal on DhcpClient so we properly started and connected the dhcpclient. We get the ip from ``get_local_interfaces`` and I am not sure what could cause this. I really would like to see the output of blueman-mechanism when connecting as it will print "bound to {someip}".

It may just be timing due to the 1ms timeout. I'll "undraft" my PR for that and also create one for stable.

btw, I am not able to reproduce here on my gentoo machine.

cschramm, 2020-02-16:
> We get the ip from `get_local_interfaces`

That's only true for master. The relevant code differs significantly in 2.1.2.

baybal, 2020-02-16:
Still no log output from blueman-mechanism, but I have something new in the syslog:

```
Feb 16 20:54:56 xps dbus-daemon[2759]: [system] Activating service name='org.blueman.Mechanism' requested by ':1.31' (uid=1000 pid=5889 comm="/usr/bin/python3.6 /usr/bin/blueman-applet ") (using servicehelper)
Feb 16 20:54:57 xps dbus-daemon[2759]: [system] Successfully activated service 'org.blueman.Mechanism'
Feb 16 20:54:57 xps dbus-daemon[5696]: [session uid=1000 pid=5694] Activating service name='org.bluez.obex' requested by ':1.43' (uid=1000 pid=5889 comm="/usr/bin/python3.6 /usr/bin/blueman-applet ")
Feb 16 20:54:57 xps obexd[6068]: OBEX daemon 5.52
Feb 16 20:54:57 xps dbus-daemon[5696]: [session uid=1000 pid=5694] Successfully activated service 'org.bluez.obex'
Feb 16 20:54:57 xps kernel: Bluetooth: RFCOMM TTY layer initialized
Feb 16 20:54:57 xps kernel: Bluetooth: RFCOMM socket layer initialized
Feb 16 20:54:57 xps kernel: Bluetooth: RFCOMM ver 1.11
Feb 16 20:55:03 xps dhcpcd[3006]: wlan0: no IPv6 Routers available
Feb 16 20:55:08 xps blueman-mechanism[6116]: INFO blueman-mechanism:68 <module>: Enabled verbose output
Feb 16 20:55:08 xps blueman-mechanism[6116]: INFO blueman-mechanism:76 <module>: Starting blueman-mechanism
Feb 16 20:55:08 xps blueman-mechanism[6116]: INFO blueman-mechanism:146 __init__: loading Ppp
Feb 16 20:55:08 xps blueman-mechanism[6116]: INFO blueman-mechanism:146 __init__: loading Network
Feb 16 20:55:08 xps blueman-mechanism[6116]: INFO blueman-mechanism:146 __init__: loading Rfcomm
Feb 16 20:55:08 xps blueman-mechanism[6116]: INFO blueman-mechanism:146 __init__: loading RfKill
Feb 16 20:55:16 xps bluetoothd[2789]:  connected
Feb 16 20:55:16 xps dhcpcd[7563]: sending commands to master dhcpcd process
Feb 16 20:55:16 xps dhcpcd[3006]: control command: /sbin/dhcpcd -m 100 bnep0
Feb 16 20:55:16 xps dhcpcd[3006]: bnep0: carrier acquired
Feb 16 20:55:16 xps dhcpcd[3006]: bnep0: IAID 96:6e:b1:11
Feb 16 20:55:16 xps dhcpcd[3006]: bnep0: adding address fe80::325c:b1cf:ef02:7f13
Feb 16 20:55:16 xps avahi-daemon[3061]: Joining mDNS multicast group on interface bnep0.IPv6 with address fe80::325c:b1cf:ef02:7f13.
Feb 16 20:55:16 xps avahi-daemon[3061]: New relevant interface bnep0.IPv6 for mDNS.
Feb 16 20:55:16 xps avahi-daemon[3061]: Registering new address record for fe80::325c:b1cf:ef02:7f13 on bnep0.*.
Feb 16 20:55:16 xps dhcpcd[3006]: bnep0: soliciting an IPv6 router
Feb 16 20:55:16 xps dhcpcd[3006]: bnep0: soliciting a DHCP lease
Feb 16 20:55:18 xps avahi-daemon[3061]: Registering new address record for fe80::de71:96ff:fe6e:b111 on bnep0.*.
Feb 16 20:55:18 xps blueman-mechanism[6045]: ERROR blueman-mechanism:41 write: Traceback (most recent call last):
Feb 16 20:55:18 xps blueman-mechanism[6045]: ERROR blueman-mechanism:41 write:   File "/usr/lib64/python3.6/site-packages/blueman/main/DhcpClient.py", line 64, in complete
Feb 16 20:55:18 xps blueman-mechanism[6045]: ERROR blueman-mechanism:41 write: ip = netifs[self._interface][0]
Feb 16 20:55:18 xps blueman-mechanism[6045]: ERROR blueman-mechanism:41 write: KeyError
Feb 16 20:55:18 xps blueman-mechanism[6045]: ERROR blueman-mechanism:41 write: :
Feb 16 20:55:18 xps blueman-mechanism[6045]: ERROR blueman-mechanism:41 write: 'bnep0'
Feb 16 20:55:21 xps dbus-daemon[5696]: [session uid=1000 pid=5694] Activating service name='ca.desrt.dconf' requested by ':1.43' (uid=1000 pid=5889 comm="/usr/bin/python3.6 /usr/bin/blueman-applet ")
Feb 16 20:55:21 xps dbus-daemon[5696]: [session uid=1000 pid=5694] Successfully activated service 'ca.desrt.dconf'
Feb 16 20:55:23 xps bluetoothd[2789]: bt_uhid_send: Invalid argument (22)
Feb 16 20:55:23 xps bluetoothd[2789]: bt_uhid_send: Invalid argument (22)
Feb 16 20:55:23 xps kernel: input: RAPOO BT4.0 MS as /devices/virtual/misc/uhid/0005:1235:AA22.0002/input/input22
Feb 16 20:55:23 xps kernel: hid-generic 0005:1235:AA22.0002: input,hidraw0: BLUETOOTH HID v0.01 Mouse [RAPOO BT4.0 MS] on DC:71:96:6E:B1:11
Feb 16 20:55:29 xps dhcpcd[3006]: bnep0: no IPv6 Routers available
Feb 16 20:55:46 xps blueman-mechanism[6045]: WARNING DhcpClient:56 _on_timeout: Timeout reached, terminating DHCP client
Feb 16 20:56:11 xps gnome-keyring-daemon[5682]: asked to register item /org/freedesktop/secrets/collection/login/16, but it's already registered
Feb 16 20:56:58 xps sudo[14268]:   baybal : TTY=pts/1 ; PWD=/home/baybal ; USER=root ; COMMAND=/bin/tail -n 100 /var/log/messages
Feb 16 20:56:58 xps sudo[14268]: pam_unix(sudo:session): session opened for user root by (uid=0)
```

baybal, 2020-02-16:
Sorry, it seems I do have an output from the blueman-mechanism

```
blueman-mechanism 21.07.15 INFO     blueman-mechanism:68 <module>  : Enabled verbose output
blueman-mechanism 21.07.15 INFO     blueman-mechanism:76 <module>  : Starting blueman-mechanism
blueman-mechanism 21.07.16 INFO     blueman-mechanism:146 __init__  : loading Ppp
blueman-mechanism 21.07.16 INFO     blueman-mechanism:146 __init__  : loading Network
blueman-mechanism 21.07.16 INFO     blueman-mechanism:146 __init__  : loading Rfcomm
blueman-mechanism 21.07.16 INFO     blueman-mechanism:146 __init__  : loading RfKill
blueman-mechanism 21.07.21 DEBUG    blueman-mechanism:163 confirm_authorization: (True, False, {})
sending commands to master dhcpcd process
blueman-mechanism 21.07.23 ERROR    blueman-mechanism:41 write     : Traceback (most recent call last):
blueman-mechanism 21.07.23 ERROR    blueman-mechanism:41 write     :   File "/usr/lib64/python3.6/site-packages/blueman/main/DhcpClient.py", line 64, in complete
blueman-mechanism 21.07.23 ERROR    blueman-mechanism:41 write     : ip = netifs[self._interface][0]
blueman-mechanism 21.07.23 ERROR    blueman-mechanism:41 write     : KeyError
blueman-mechanism 21.07.23 ERROR    blueman-mechanism:41 write     : :
blueman-mechanism 21.07.23 ERROR    blueman-mechanism:41 write     : 'bnep0'
blueman-mechanism 21.07.51 WARNING  DhcpClient:56 _on_timeout: Timeout reached, terminating DHCP client
```

baybal, 2020-02-16:
```
blueman-mechanism 21.20.21 DEBUG    blueman-mechanism:163 confirm_authorization: (True, False, {})
sending commands to master dhcpcd process
blueman-mechanism 21.20.22 INFO     DhcpClient:62 _check_client: netifs
blueman-mechanism 21.20.22 INFO     DhcpClient:63 _check_client: {'lo': ('127.0.0.1', '255.0.0.0'), 'wlan0': ('10.79.105.31', '255.255.224.0')}
blueman-mechanism 21.20.22 INFO     DhcpClient:64 _check_client: self
blueman-mechanism 21.20.22 INFO     DhcpClient:65 _check_client: <DhcpClient.DhcpClient object at 0x7f394751d8b8 (blueman+main+DhcpClient+DhcpClient at 0x56139f15e0c0)>
blueman-mechanism 21.20.22 INFO     DhcpClient:66 _check_client: iface
blueman-mechanism 21.20.22 INFO     DhcpClient:67 _check_client: bnep0
blueman-mechanism 21.20.23 ERROR    blueman-mechanism:41 write     : Traceback (most recent call last):
blueman-mechanism 21.20.23 ERROR    blueman-mechanism:41 write     :   File "/usr/lib64/python3.6/site-packages/blueman/main/DhcpClient.py", line 70, in complete
blueman-mechanism 21.20.23 ERROR    blueman-mechanism:41 write     : ip = netifs[self._interface][0]
blueman-mechanism 21.20.23 ERROR    blueman-mechanism:41 write     : KeyError
blueman-mechanism 21.20.23 ERROR    blueman-mechanism:41 write     : :
blueman-mechanism 21.20.23 ERROR    blueman-mechanism:41 write     : 'bnep0'
blueman-mechanism 21.20.51 WARNING  DhcpClient:56 _on_timeout: Timeout reached, terminating DHCP client
```

baybal, 2020-02-16:
get_local_interfaces doesn't return bnep0 for some reason. I did a big python update recently. Not sure what exact package might be responsible for that.

baybal, 2020-02-16:
siocgifconf will not give you bnep0 if it has no ip address at all

baybal, 2020-02-16:
Clearly something is going on with dhcpcd. It fails to get a V4 address.

baybal, 2020-02-16:
I can confirm now, it's dhcpcd itself having troubles getting a lease.

I tried with dhclient, and it worked just fine

infirit, 2020-02-23:
I think the actual problem is that we hit the timeout of 120 ms, bnep0 is torn down and dhcpcd exits by itself. I fixed the timeout to be 120 seconds in #1183 (2.1) and #1177 (master).

I also tried dhclient and for me it just keeps running ignoring what happens to bnep0. Even with the correct timeout I end up with as many dhclient instances as the number of attempts to connect. @cschramm it wold be nice if you can confirm this with dhclient.

I would appreciate if you could try the change and confirm if dhcpcd still causes a problem. If not its fine as well then I'll close this issue when the 2 PR are merged.

closed 2020-07-19 by stale[bot]