blueman-project/blueman #1178: I cannot connect to Internet on my device from another one

geremi1, 2020-02-11:
blueman: 2.1.1-2
BlueZ: 5.52-2 & bluez-hid2hci 5.52-2
Distribution: Arch Linux Latest
Desktop environment: Compiz mix KDE

* [ Yes] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Ok. My friend is having a problem. When he connects to the all working bluetooth computer of mine, everything is working, except the bluetooth network connection. It shows as "paired" on the laptop and the iPhone, but no internet connection seems to exist. It worked is the past, but doesn't anymore. 

Here is the log of blueman :  "https://pastebin.com/fZ5LE5ch"

How may I get it working ?

cschramm, 2020-02-11:
I suspect an issue with the DHCP configuration.
Please provide your networking service configuration, best by providing `/var/lib/blueman/network.state`.
The logs for `blueman-mechnism.service` and its subprocesses might also yield some hints.

geremi1, 2020-02-12:
- Here is var/lib/blueman/network.state :

cblueman.main.NetConf
NetConf
q)q}q(XversionqK
X
 dhcp_handlerqcblueman.main.NetConf
DnsMasqHandler
q)q}q(XpidMXQXnetconfq	hubX	ipt_rulesq
]q
  (Xnatq
        X
X+-s 10.245.181.1/255.255.255.0 -j MASQUERADEqqXfilterqXFORWARDqX-i pan1 -j ACCEPTqqhhX-o pan1 -j ACCEPTqqhhhqeX
                                                                                                                ip4_addressqX
255.255.255.0qX                                                                                                              10.245.181.1qip4_maskqX
               ip4_changedqb.

- Here is blueman-mechanism.service : https://pastebin.com/3YZcke9X

cschramm, 2020-02-13:
DHCP looks fine. In the log the device gets address 10.245.181.144. Can you ping it? Maybe you just do not have any routing configured? If you see the device's traffic on `pan1` e.g. with tcpdump / Wireshark, the setup is fine.

geremi1, 2020-02-13:
- I can ping it and I receive the packets with the following command : "ping 10.245.181.144".
- I can also see some traffic on pan1 when connecting my device : https://pastebin.com/vjBKCtNk
- This is my routing configuration (ip route) : 

default via 10.1.80.1 dev wlo1 proto dhcp metric 600
10.1.80.0/20 dev wlo1 proto kernel scope link src 10.1.85.131 metric 600
10.245.181.0/24 dev pan1 proto kernel scope link src 10.245.181.1 metric 425

is this all normal ? It indicates that I'm connected, but not to the Internet

cschramm, 2020-02-15:
The network looks good to me. Things like setting up a NAT are out of blueman's scope.

I you want to enable 10.245.181.0/24 to connect to the internet via a router at 10.1.80.1, you at least need to enable ip forwarding and either add routes for 10.245.181.0/24 to 10.1.80.1 or set up some NAT on your host, see e. g. https://netfilter.org/documentation/HOWTO//NAT-HOWTO.html.

closed 2020-02-15 by cschramm