blueman-project/blueman #1179: Build is getting failed with error code 2=“null” and 255=“null” in Windows 10 OS

paratsa, 2020-02-13:
Greetings,

We have recently migrated to Windows 10 from Windows 7.

When we are building our code in Windows 10, we are getting the build error such as 255="null" and 2="null".

We have no clue how to fix this build error.

We are facing this build error after we migrated to Windows 10.

Our Code was successfully built when we used Windows 7 OS.

Please see the build error in the below mentioned log.

Error:

**2020-02-12 18:19:46 t-58 WARNING: Error: 255="null" when executing[C:\Clones\Exp\TSP_VOB\Build\bin\bldcomp.bat, libVAL, OSE]**

**2020-01-09 08:50:55 t-52 WARNING: Error: 2="null" when executing[G:\TSP_VOB\Build\bin\bldcomp.bat, libVAL, OSE]**

Here bldcomp.bat is the batch file which will invoke the make file the make file inturn will invoke the make.exe,libVAL is the component to be build and the OSE is the configuration for which the build is done.

Here bldcomp.bat, libVAL and OSE is fed to the processbuilder java process as inputs, while the build is going on the build(processbuilder) is interrupted with the error code 2="null" and 255="null"

Please see the environment details below.

Rhapsody 8.0.5
Java Version 7
Visual studio 2005
gcc version 4.2.3
Windows Version : Windows 10    

Note :

1)The configuration can be of any specific target such as OSE(arm),WinXP(Desktop simulation),WinCE and Lint.

2)The Component can be Utilities,SIM,OSAL,ARCOM,FD,VAL,TRF,SVAL and MSW

3)The build error is appearing in any component and in any configuration randomly(meaning issue place is not definite).

4)This build went through fine with out any issues on Windows 7 OS, now we are facing this build errors in Windows 10 OS.

Please help us to fix this issue.

Thanks in advance,

S.Parthasarathi

closed 2020-02-13 by cschramm

paratsa, 2020-02-13:
Why did you close this query ?

cschramm, 2020-02-13:
Why did you open this? This is the blueman project and you do not refer to it at all.