blueman-project/blueman #1184: Bluetooth PAN using connman

baybal, 2020-02-17:
Hello,

Connman has a network manager emulation layer. Can we add support for it for PAN connections, or use connman's native API?

Connman picks up PAN connections just fine, but you need to open a PAN connection first.

infirit, 2020-02-23:
We don't use networkmanager for this but if someone want to add this they can. marking as feature request.

infirit, 2020-02-23:
PAN, NAP... I always get them confused, I blame bluetooth for their horrible naming conventions :smiling_imp:. So yes, there is a plugin for networkmanager and it uses the gi bindings to communicate with NM. 

I hate working with connman so if someone else wants to work on this have fun :smiley:

baybal, 2020-02-27:
> there is a plugin for networkmanager and it uses the gi bindings to communicate with NM.

Can you check if connman's compatibility layer for NM supports same dbus calls?

infirit, 2020-03-05:
I could not find anything like nm compat layer. Also I would prefer it to
be its own plugin for blueman applet and use connman's own dbus API.

baybal, 2020-03-05:
@infirit it seems it was removed many versions ago

>  * Remove NM compatibility mode from default init-script options (no longer
>    available upstream); remove connman.default accordingly

baybal, 2020-03-05:
So yeah, no way around using its own API now