blueman-project/blueman #1188: Bluetooth setup assistant has control with missing borders

philipzae, 2020-03-04:
blueman: 2.1.2
BlueZ: 5.53
Distribution: Xubuntu 20.04
Desktop environment: XFCE

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The bluetooth assistant has a steps control on the left but unfortunately only 1 of the 4 borders appear.

[image]

Also filed at launchpad
https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/1866028

infirit, 2020-03-05:
Can you reproduce this with the gtk default theme adwaita? If not then it
is most likely a problem with the theme. If it does we'll have to check if
I see the same

philipzae, 2020-03-06:
Yes its there with adwaita.

[image]

cschramm, 2020-03-06:
I think that's just what [GtkAssistant](https://developer.gnome.org/gtk3/stable/GtkAssistant.html) does when using defaults (`use-header-bar` `FALSE` and `has-padding` `TRUE`).

infirit, 2020-03-11:
>
> I think that's just what GtkAssistant
> <https://developer.gnome.org/gtk3/stable/GtkAssistant.html> does when
> using defaults (use-header-bar FALSE and has-padding TRUE
>
It is probably how GtkAssistant is designed and not much we can do about it
then. I'll poke at some properties to see if it makes a difference.

>

philipzae, 2020-05-11:
@infirit any progress with this issue?

infirit, 2020-05-11:
Yes, this is how it looks. Nothing we can do about this.

closed 2020-05-11 by infirit