blueman-project/blueman #1189: I can’t turn on the PAN access point

MCWayman, 2020-03-11:
blueman 2.0.5 ubuntu Ubuntu 18.04.4

I can’t turn on the PAN access point
When you try to enable the NAP network access point in local blueman services appears "Failed to apply network settings", and exception:
org.freedesktop.DBus.Python.FileNotFoundError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 36, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python3/dist-packages/blueman/plugins/mechanism/Network.py", line 59, in EnableNetwork
    nc.apply_settings()
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 311, in apply_settings
    ret = call(["ifconfig", "pan1", ip_str, "netmask", mask_str, "up"])
  File "/usr/lib/python3.6/subprocess.py", line 287, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'ifconfig': 'ifconfig'

I would be grateful for your help. Sorry for my english.

infirit, 2020-03-11:
That (old) version of blueman still relies on net-tools package so install
that and it should solve the error.

MCWayman, 2020-03-11:
Sorry, but I did not understand what needs to be installed?

infirit, 2020-03-11:
net-tools

MCWayman, 2020-03-12:
It really worked. Thanks.

closed 2020-03-12 by MCWayman