blueman-project/blueman #1191: turn off permanently on aarch64

ng0177, 2020-03-15:
Sorry, there seems to be no way of doing so. Ideas? Thanks.

cschramm, 2020-03-15:
I don't think there's anything specific to aarch64. There are a couple of mechanism that might be auto-powering your adapters, e.g. udev rules, BlueZ' `AutoEnable`, blueman's `auto-power-on`...

ng0177, 2020-03-15:
I tried `tlp` but it seems to have no effect. How test on cli, if bluetooth is really on or off?

cschramm, 2020-03-15:
`hciconfig` will either show `UP` or `DOWN` for the controllers.

ng0177, 2020-03-15:
```
Building bluez-hciconfig...
==> ERROR: bluez-hciconfig is not available for the 'aarch64' architecture.
```

cschramm, 2020-03-15:
BlueZ' D-Bus interface would be what I'd use then, e.g.
```
dbus-send --print-reply --system --dest=org.bluez /org/bluez/hci0 org.freedesktop.DBus.Properties.Get string:org.bluez.Adapter1 string:Powered
```
in case `dbus-send` is available. :smile: Some other tool otherwise...

ng0177, 2020-03-15:
```
method return time=1584263958.358546 sender=:1.18 -> destination=:1.212 serial=138 reply_serial=2
   variant       boolean true
```
works like a charm :-) Can `send-bus` also be used to turn it on and off?

cschramm, 2020-03-15:
```
dbus-send --print-reply --system --dest=org.bluez /org/bluez/hci0 org.freedesktop.DBus.Properties.Set string:org.bluez.Adapter1 string:Powered variant:boolean:false
```

closed 2020-03-15 by cschramm

ng0177, 2020-03-15:
Problem solved. Thanks for the swift responses!