blueman-project/blueman #1204: Gtk.Menu.popup() is deprecated and breaks on Wayland

Xenopathic, 2020-03-29:
The method `popup()` of `Gtk.Menu` is deprecated since GTK 3.22 (https://lazka.github.io/pgi-docs/#Gtk-3.0/classes/Menu.html#Gtk.Menu.popup) (https://developer.gnome.org/gtk3/stable/GtkMenu.html#gtk-menu-popup), and does not work on Wayland compositors such as Sway. It should be replaced by `popup_at_widget()`, `popup_at_pointer()` or `popup_at_rect()`.

Other GTK projects have also encountered this, e.g. [Nautilus](https://bugzilla.gnome.org/show_bug.cgi?id=772096).

Also see https://github.com/swaywm/sway/issues/3135

infirit, 2020-03-31:
Merged  #1205 but would be nice if you could confirm it's working now.

Xenopathic, 2020-04-06:
Confirmed, it now works \o/

closed 2020-04-06 by Xenopathic