blueman-project/blueman #1208: DBusService or DBusProxy broken on glib 2.64

infirit, 2020-04-05:
It register and owns the name fine but we then block and timeout. It may be that we don't get a proxy for org.bluez. I downgraded back to 2.62 and things started working again.

Debian testing and unstable also have glib 2.64 so it would be nice to know if this is just my gentoo install or also show up in debian. @cschramm if you have the means could you check on your machine or the vm you created recently?

cschramm, 2020-04-06:
How to replicate? I do not seem to have any issues when running 2.1.2 on 2.64.

cschramm, 2020-04-06:
Got it in master. 2.1.2 is actually fine.

> c0335308d4a341d87fd9ee8d5b81ab91fad0b307 is the first bad commit

cschramm, 2020-04-06:
> 1cbd175f9d28a8af257d8e55a6f882ba1c9e8442 is the first bad commit

cschramm, 2020-04-06:
https://github.com/blueman-project/blueman/commit/1cbd175f9d28a8af257d8e55a6f882ba1c9e8442#diff-4519b7078a6f26a78abee36139b57055R93 does not terminate on 2.64.

infirit, 2020-04-06:
> Got it in master. 2.1.2 is actually fine.

Sorry, should have mentioned this :fearful: 

> [1cbd175#diff-4519b7078a6f26a78abee36139b57055R93](https://github.com/blueman-project/blueman/commit/1cbd175f9d28a8af257d8e55a6f882ba1c9e8442#diff-4519b7078a6f26a78abee36139b57055R93) does not terminate on 2.64.

I also tried read_chars with the same result. I tried that as it does not have an pygobject override like read does.

Thanks for confirming, updated the title.

I'll open an issue with pygobject with a minimal example and see where it leads me. You want me to revert or wait for some feedback from pygobject/glib?

cschramm, 2020-04-06:
I seem to be too stupid for gdb today but what I've seen so far (primary with `strace`) is that when `g_io_channel_read_chars` gets called from the `_gi` extension, 2.64 does multiple `read` syscalls while 2.62 just does one and then returns. I can not replicate that with C code, only via the `_gi` Python extension.

cschramm, 2020-04-06:
https://gitlab.gnome.org/GNOME/glib/-/blob/glib-2-64/glib/giochannel.c#L1591 returns `-2` instead of `0` due to https://gitlab.gnome.org/GNOME/glib/-/commit/568720006cd1da3390c239915337ed0a56a23f2e. `g_io_channel_fill_buffer` returns `G_IO_STATUS_NORMAL` nonetheless and sends `g_io_channel_read_chars` into a loop.

cschramm, 2020-04-06:
Got a fix in #1211. We need to override the default UTF-8 encoding to read binary data. A loop in `g_io_channel_read_chars` is obviously not what should happen on `U+0000` but we should not read UTF-8 at all anyway.

infirit, 2020-04-07:
Setting encoding makes no difference for me. Minimal example that blocks for me is below. I get exactly one "Alive" and then nothing.
```python
from gi.repository import GLib

def tick():
    print("Alive")

def on_event(channel, condition):
    print(channel, condition)

channel = GLib.IOChannel.new_file("/dev/rfkill", "r")
channel.set_encoding(None)
wid = channel.add_watch(GLib.IO_IN | GLib.IO_ERR | GLib.IO_HUP, on_event)

GLib.timeout_add_seconds(1, tick)

loop = GLib.MainLoop()
loop.run()
```

PR #1211 (2020-04-07): Fix IOChannel encoding
Closes #1208
commit 833a8085 (2020-04-07): Fix IOChannel encoding

Closes #1208
closed 2020-04-07 by infirit