blueman-project/blueman #1209: Sony WH-1000XM3 connection on Debian 10 Buster

Phil995511, 2020-04-06:
blueman: 2.0.8-1
BlueZ: 5.50-1.2~deb10u1
Distribution: Debian 10 Buster x64 Bits (AMD64)
Desktop environment: Cinnamon 3.8

Hello, I'm using my new Linux Debian 10 Buster headphones. 

When I link it to my laptop everything works perfectly the first time (although I unfortunately do not seem to benefit from the LDAC codec for which I bought this headset, it works satisfactorily in A2DP Sink). If I turn off this headset and I want to reuse it later, the connection is made but instead of being of good quality (A2DP Sink) it is done in SBS (HSP / HFP) which is of very poor quality. I am very unhappy with the result... 

Is there a way to force the connection in A2DP Sink or to deactivate the codec SBS (HSP / HFP) which I do not use and thus to force the connection in A2DP mode Sink ? Is it possible to use the LDAC codec from Sony which is available for free for Android phones which is based on Linux ? 

Best regards.

infirit, 2020-04-07:
What codec is used it up to pulseaudio and blueman has no control over it. From blueman and most audio mixer you can select A2DP again.

If you want LDAC you need different pulseaudio bluetooth modules, see https://github.com/EHfive/pulseaudio-modules-bt

Phil995511, 2020-04-08:
Hello, I cannot select A2DP again, it remains blocked in SBS (HSP / HFP). I have to turn off the headphones and reconnect manually, which is very annoying in the long run.

I will change OS if I do not find a viable solution...

I found the package "libldac_2.0.2.3-2_bionic1_amd64.deb" which is precompiled for kernel 4.19, it installs without difficulties but it is impossible to make it work with other softwares to enjoy sound in LDAC quality ;(

Regards.

hugokerstens, 2020-05-06:
@Phil995511 What works for me as well is disconnecting the headset from the blueman manager and reconnect as audio sink, no need to turn off the headset itself.

This issue has been reported multiple times: #1031 #1242

Phil995511, 2020-05-07:
Hello Hugo Kerstens,

Thank you for your good advice, from memory it seems to me to have already tested this idea without success with each reboot I have the same problems...

Debian 10.4 comes out on Saturday 07.05.2020, I will reinstall my laptop on this occasion on an external disk to repeat the test according to your advice.

For the moment the only solution I have found is using Ubuntu 20.04 for fix this problem... Unfortunately under Debian, I can not use the double GPU of my laptop either, none tutorial for the installer of Bumblebee / Nvidia Optimus works ;-(

Regards

> @Phil995511 What works for me as well is disconnecting the headset from the blueman manager and reconnect as audio sink, no need to turn off the headset itself.
> 
> This issue has been reported multiple times: #1031 #1242

closed 2020-05-08 by cschramm

Phil995511, 2020-05-10:
@hugokerstens The problem persist, if you reconnect as audio sink (or with an other option), then you disconnect and reconnect the Sony WH-1000XM3, you are on "HSP/HFP" codec with horrible audio quality not in "A2DP Sink" codec. 

@cschramm Please dont close prematurely tickets. The problem is not solved, Please solve it...

cschramm, 2020-05-10:
@Phil995511: To my knowledge, BlueZ 5.51 brings a solution for this (pointed out by @hugokerstens). Feel free to continue the thread but it's nothing we need to track from the blueman perspective, thus closed.

Phil995511, 2020-05-10:
@cschramm Thank you for this information, in this case, the Debian people should update BlueZ on version 5.51 !! Can you give them a request to this effect, to prevent other users from being stuck with this problem ?? Regards.

cschramm, 2020-05-10:
@Phil995511: The way to ask for that would be to [report a bug](https://wiki.debian.org/HowtoUseBTS#Reporting_bug_reports:_reportbug) against the bluez package, however, rest assured that the Debian maintainers in general are well aware of updating packages to new release versions unless the package is orphaned (which bluez is not) in which case nobody would be there to handle the request. Also note that bluez 5.52 is packaged and available in the experimental repository, so that your request would be rather pointless, and as you seem to use a stable release you would not receive any possible update anyway. The way to go for a stable release would be a backport. You might consider asking for one at [the respective mailing list](https://lists.debian.org/debian-backports/) but I'd expect the community to respond with "go ahead, start working on it" 😉.

Phil995511, 2020-05-10:
@cschramm As you can see below for example the version 10.4 of Debian fixes many bugs but, but not that of Bluez ;-(

https://www.debian.org/News/2020/20200509

cschramm, 2020-05-10:
@Phil995511: As I wrote, the Debian maintainers *are* aware of the fact that new versions get released and their packages need updates, there is *no* need to remind them and it will *not* make them do their volunteer work faster for you, a newer version of BlueZ than 5.50 *is* available in the archives, and will *not* be brought to an existing stable release that you're using.

Also, I'd hardly consider the behavior you're experiencing to be a bug and if it is one at all, it is a bug of your device, so please complain to Sony who you paid for their stuff and not the communities behind BlueZ, Debian, or blueman. People at BlueZ even improved your life with those devices (committed 06/2018 (kudos to @Vudentz), first released 09/2019, packaged in Debian (kudos to @iwamatsu) just two weeks later).