blueman-project/blueman #1210: blueman-tray crashed with FileNotFoundError in check_single_instance()

datyuesh, 2020-04-06:
blueman:2.1.2-1
BlueZ:5.53-0ubuntu2
ExecutablePath:/usr/bin/blueman-tray
ApportVersion:2.20.11-Oubuntu22
Architecture:amd64
Uname:Linux5.4.0-21-generic x86_64

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

datyuesh, 2020-04-06:
I have two environments, laptop and desktop

This error is reported every time the desktop computer is turned on. I think it may be related to that my desktop computer does not have Bluetooth device, but today there is also this problem in my notebook. As soon as I turn on the computer, there is an error reporting pop-up window.The error content is the same

  Traceback:
   Traceback (most recent call last):
     File "/usr/bin/blueman-tray", line 40, in <module>
       BluemanTray()
     File "/usr/lib/python3/dist-packages/blueman/main/Tray.py", line 14, in __init__
       check_single_instance("blueman-tray")
     File "/usr/lib/python3/dist-packages/blueman/Functions.py", line 306, in check_single_instance
       os.remove(lockfile)
   FileNotFoundError: [Errno 2] No such file or directory '/home/kylin/.cache/blueman-tray-1000'
  UserGroups: adm cdrom dip lpadmin lxd plugdev sambashare sudo 
  _LogindSession: c2
  ApportVersion: 2.20.11-0ubuntu22
  Architecture: amd64

infirit, 2020-04-08:
We should not error here but I do wonder how you got to it. The thing is that a few lines earlier in the code we checked that the file exists. There must be something that deleted it between checking existence and deleting it.

The error is harmless so you should not worry, it can be annoying. I am not at my workstation I do development on but when I am I'll probably ask for some test to run.

infirit, 2020-04-08:
I fixed the immediate problem of the error with PR #1212, please try. But I would like to understand how you got into this situation so please read below.

Can you post the output of ``ls -la ~/.cache/blueman-tray-1000``. Also can you post blueman-applet debug log as described in https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

datyuesh, 2020-04-09:
`ls -la ~/.cache/blueman-tray-1000`
-rw-rw-r--  1  blue  blue  6  April     9  09:28  blueman-tray-1000

datyuesh, 2020-04-09:
Thank you for the patch.
Before that, I wrote a patch and there is no error report at present

 From 00229e8451ec559da8f349771c3b2a213ba6e2a0 Mon Sep 17 00:00:00 2001
| From: Yue Shuai <yueshuai@kylinos.cn>
| Date: Tue, 7 Apr 2020 17:10:07 +0800
| Subject: [PATCH] avoid Ubuntu's apport error capture
   
  Signed-off-by: Yue Shuai <yueshuai@kylinos.cn>
  ---
   blueman/Functions.py | 7 +++++--
   1 file changed, 5 insertions(+), 2 deletions(-)
   
- diff --git a/blueman/Functions.py b/blueman/Functions.py
| index 3b76271e..aec72137 100644
| --- a/blueman/Functions.py
| +++ b/blueman/Functions.py
| @@ -302,8 +302,11 @@ def check_single_instance(name, unhide_func=None):
|          pid = get_pid(lockfile)
|          if pid:
|              if not is_running(name, pid):
| -                print("Stale PID, overwriting")
| -                os.remove(lockfile)
| +                try:
| +                    print("Stale PID, overwriting")
| +                    os.remove(lockfile)
| +                except:
| +                    print("Remove lockfile failed")
|              else:
|                  print("There is an instance already running")
|                  time = os.getenv("BLUEMAN_EVENT_TIME") or 0

datyuesh, 2020-04-09:
I tested two systems, the latest version of Ubuntu and Ubuntu-kylin

Desktop computer with no Bluetooth device(100%Appear)
Laptop with Bluetooth device(Rarely appear)

Ubuntu: Blueman is not integrated by default,so i try `sudo apt-get install blueman` and reboot.
              One turn on, Ubuntu's apport will pop up the window and report an error `blueman-tray crashed with FileNotFoundError in check_single_instance()`

Ubuntu-kylin:Blueman is integrated by default.  One turn on, Ubuntu's apport will pop up the window and report an error `blueman-tray crashed with FileNotFoundError in check_single_instance()`

When blueman is installed by default, there is pop-up error every time you power on.
It's really annoying because I don't have a Bluetooth device on my desktop, and the service of both BlueZ and Blueman is dead

PR #1212 (2020-04-28): Handle os.remove failing in check_single_instance
fixes #1210
commit f71e3926 (2020-04-28): Handle os.remove failing in check_single_instance

fixes #1210
PR #1213 (2020-04-28): Handle os.remove failing in check_single_instance
fixes #1210
commit 991c671f (2020-04-28): Handle os.remove failing in check_single_instance

fixes #1210
closed 2020-04-28 by infirit