blueman-project/blueman #1217: libappindicator deprecation

cschramm, 2020-04-15:
Debian deprecated libappindicator in favor of the Ayana project which I haven't heard of before, see [blueman bug #956761](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=956761). However, I was well aware that libappindicator is completely unmaintained, so this makes sense. I didn't have a closer look yet, so I'm unsure if it is a drop-in replacement or if the API differs.

infirit, 2020-04-15:
It looks like an drop in replacement looking at the two apis on https://lazka.github.io/pgi-docs/

However, while libayatana-appindicator get some updates it is not really looking well maintained either.

Assuming it is a real drop in replacement, my suggestion would be to try to import libayatana-appindicators and fallback onto libappindicator.

libayatana-appindicator is not packaged  in gentoo so I won't be able to test this easily.

cschramm, 2020-04-15:
Looks pretty much like a Debian-ish approach as the Ayatana stuff isn't available anywhere else (yet). I'd stick to keeping a patch for the Debian package then for the now.

infirit, 2020-05-01:
If the patch is just a try/except block around the import just open a PR :wink:

closed 2020-07-04 by cschramm

cschramm, 2020-07-04:
Won't fix for now. I added a patch to the Debian build that just replaces `AppIndicator3` with `AyatanaAppIndicator3` in `gi.require_version` calls and the `gi.repository` import, so that they could remove libappindicator but neither the Debian effort nor the Ayatana project seem very active, so I don't expect this to get relevant in other distributions.