blueman-project/blueman #1219: Unable to run blueman-manager

xXBannedXx, 2020-04-15:
blueman: latest
BlueZ: latest
Distribution: Kali
Desktop environment: Kali Linux

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Yannik, 2020-04-26:
@xXBannedXx run like this: `blueman-manager --loglevel DEBUG`

xXBannedXx, 2020-04-27:
blueman-manager 21.15.56 INFO     PluginManager:85 load_plugin: ['Notes', 'Services', 'PulseAudioProfile', 'Info']
blueman-manager 21.15.56 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.manager.Notes.Notes'>
blueman-manager 21.15.56 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.manager.Services.Services'>
blueman-manager 21.15.56 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'>
*** stack smashing detected ***: <unknown> terminated
Aborted

cschramm, 2020-04-28:
Some binary, either our C library or your Python interpreter in general trips over a GCC protection mechanism there. How do you build blueman?

xXBannedXx, 2020-04-28:
I use the default build that comes with kali Linux. Usually an apt-get update/upgrade will update it from kalis servers.

cschramm, 2020-05-14:
I'd expect that to be some Kali specific issue, not related to blueman (as they just import blueman from Debian without changes). Please report it to the Kali Linux bug tracker.

closed 2020-07-20 by stale[bot]