blueman-project/blueman #1231: blueman-manager crashes when bluetoothd is restarted

Yannik, 2020-04-26:
blueman: `1:2.1.2-1.fc31`
BlueZ: `5.54-1.fc31`
Distribution: Fedora 31
Desktop environment: gnome

When restarting bluetoothd (`systemctl restart bluetooth.service`), `blueman-manager` crashes:
```
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x90 in position 1: invalid start byte
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x90 in position 0: invalid start byte
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf9 in position 1: invalid start byte
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb0 in position 0: invalid start byte
blueman-manager 12.33.47 ERROR    Manager:118 on_dbus_name_appeared: Default adapter not found, trying first available.
blueman-manager 12.33.47 ERROR    Manager:122 on_dbus_name_appeared: No adapter(s) found, exiting
```

This is especially annoying as my initramfs is locking my encrypted luks `/` volume and chrooting into the initramfs to ensure data security. To do that, it need to `systemctl stop bluetooth` before suspending, and `systemctl start bluetooth` while resuming, because bluetoothd can otherwise inhibit the suspend process as it generates i/o on the locked encrypted root volume.

This means that blueman-manager crashes after every suspend.

infirit, 2020-04-27:
What I wanted to do (FIXME is mine) is to replace the devicelist with a frame which contains a message that either bluetooth is not available or that the adapter is not powered. In case of the latter show a button or provide instruction to power the adapter.

Yannik, 2020-04-27:
@infirit With my patch, when no adapter is available, no devices are shown and the adapte menu is greyed out. I think this is a pretty good indication that the bluetooth adapter is not available (plus there's always the stdout/stderr output) already. Adding an extra frame like you suggested seems like a nice suggestion though. 
What about opening another issue to track this?

I'd like to keep this separate because this PR is targeted at fixing the crash on stop/start.

cschramm, 2020-05-14:
I'm a little bit confused what's the actual issue here. Looking at #1235, #1245 should fix it, correct? How's #1234 related?

infirit, 2020-05-14:
It started with several exception when more than one device is removed from the list.  #1234 will sort this.

The other is that crash handlers (abrt) will pickup ``exit(1)`` and report it as a crash. In our case we really expect to exit so sending exit code 1 is a bit harsh. I did  #1245 for this.

Then the one I do not agree with is  #1235 as it basically removes the error dialog en shutdown keeping blueman-manager open. This is technically no problem but I want the user to be properly informed that bluetooth is not available/blocked, ideally with a button to power the adapter.

cschramm, 2020-05-15:
OK, so this is actually about two issues, one already resolved by #1245 and one resolved in #1234. #1235 can be closed then and this ticket can be closed when #1234 is merged.

PR #1234 (2020-05-16): Fix device removal handling
This fixes multiple issues:
- if two devices are removed within the time the fader animation takes,
  _self._faderhandler is overwritten with the faderhandler id of the latter
  fader, which leads to non-removal of the first fader handler
- if two devices are removed within the time the fader animation takes,
  _self._faderhandler is set to None by the first __on_fader_finished
  callback, leading to a crash in the second callback because
  disconnect is called as `fader.disconnect(None)`
- If /org/bluez/hci0/dev_XX is removed, and then /org/bluez/hci0 is
  removed  [... cut]
closed 2020-05-16 by infirit